ome/bioformats #3702: BUG: bfconvert sets wrong fps when exporting `.cxd` to `.avi`

rkrug, 2021-06-03:
I am using bfconvert to convert `.cxd` files to `.avi` false using the following command:

```
bfconvert -overwrite -no-upgrade input.cxd output.avi
```

even though the delay between the images taken is 0.04 (= 24 fps) and we are recording for 5 seconds, the resulting `.avi` is 12 seconds and 10 fps.

This requires using `ffmpeg` to change the fps AND keep the number of frames the same.

Am I missing something? I did not find any option to fix this?

dgault, 2021-06-04:
Hi @rkrug, thank you for opening the issue. Do you have a link to the original cxd file that we can test? If you need a location to upload the file to you can use https://zenodo.org/ 

10fps is the default value and it looks as though when going through the convert tool it is not attempting the set the fps so this would be considered a bug in the AVIWriter.

rkrug, 2021-06-04:
@dgault I have put it into my google dive for now - you can access it from https://drive.google.com/file/d/13ko1z5dGgiwUSqQ80-rh2AX8jGL2fV51/view?usp=drivesdk

Thanks for looking into this,

Rainer

dgault, 2021-09-20:
Thanks Rainer for supplying the sample file and my apologies for the lack of follow up to this issue since. I have opened a PR which will hopefully address this bug in writing AVI files for the future.

closed 2021-11-19 by melissalinkert