ome/bioformats #3703: Ventana bif issue

underscorenico, 2021-06-04:
I have found an issue when using bioformats (6.5.1) for opening Ventana Bif files. 

More specifically I used QuPath (0.2.3 version which embeds bioformats 6.5.1) to open the OS-2.bif file, from public dataset images available at the openslide website (http://openslide.cs.cmu.edu/download/openslide-testdata/Ventana/OS-2.bif).

*Note: the VentanaReader class has not changed since the v6.5.1.*

The problem I see is that regularly I see stitching issues. Here are two examples:
[image]
[image]

The first one is at X pixel 41653 and the second one is at pixel 42564. The interval between these two artifacts is the same across the entire image and is 911 pixels. 

Now that number (911 pixels) is coherent with what I see in the code: an average of the X overlap is computed (when debugging I got the number 113 for the variable `rightSum`) and since the tile width of this image is 1024, then it means that every tile is affected by substracting 113 pixels from it. This is exactly the artifact we are observing.

I get that this behaviour is not an accident since it is explained in the comments:

```
      // list of overlaps is not authoritative
      // some values will be wrong, and some overlaps will be missing
      // average the RIGHT values and apply the average to all tiles
      // then average the UP values and apply to all tiles
```

But I would like to understand why you decided to average the overlaps that are present in the file's metadata and substract them from *every* tile in the image, instead of simply substracting the overlaps from the tiles that are specified in the metadata.

For example I from the `tiffinfo` output of the OS-2.bif file:

```
...
<TileJointInfo FlagJoined='1' Confidence='99' Direction='RIGHT' Tile1='20' Tile2='19' OverlapX='115' OverlapY='1'/>
...
```
So that information (`Tile1` and `Tile2`) can be used instead of computing an average, right?

Thank you very much for your time and work on this project :)

melissalinkert, 2021-06-07:
Seamless stitching is definitely not expected for any Ventana data.  Our experience with a range of .bif files indicates that the tile overlaps in the metadata are not fully representative of the how the tiles should actually be pieced together.  Sometimes tiles that clearly overlap do not have a ```TileJointInfo``` entry, and sometimes that entry is clearly incorrect.  The averaging strategy currently used so far provides the least-bad results, given that sophisticated stitching is outside the scope of Bio-Formats.

If you are using Bio-Formats in other contexts (direct API access, command line tools, etc.), then setting the ```ventana.split_tiles``` option to ```true``` would allow you to retrieve the unstitched source tiles.  You can then use existing stitching tools or your own algorithm.  I am not aware of a way to set this option in QuPath, but you can find additional information here:

https://docs.openmicroscopy.org/bio-formats/6.6.1/formats/options.html#usage

You can also easily compare how OpenSlide handles this data from QuPath, by unchecking ```Enable Bio-Formats``` in the ```Preferences``` window and re-opening the image.

melissalinkert, 2023-01-03:
Closing as there isn't anything to be fixed here. Happy to continue the conversation though if there are any additional questions.

closed 2023-01-03 by melissalinkert