ome/bioformats #3704: How to get label image or thumbnail Image from  tiff file ?

yaokkkkkk, 2021-06-10:
some tilff files contains label image and thumbnail image,how to get them ,please
https://openslide.org/formats/philips/

dgault, 2021-06-10:
Hi @yaokkkkkk, if the labels and thumbnail images are included they will appear as additional series on the image. How are you accessing the images? If you are using the Bio-Formats API directly you can set the series to read as in this example: https://github.com/ome/bioformats/blob/develop/components/formats-gpl/utils/ConvertToOmeTiff.java#L64

yaokkkkkk, 2021-06-10:
Thanks very much. I used to use openslide , and now I use bioformats. Set the series means that I should know the name of the series or image, I know that I can use the method **getImageName()** to get the name. But why not provide a convenient way to get it? It's just advice. Thank you very much for your answering.

sbesson, 2021-06-14:
@yaokkkkkk thanks for the feedback. I think the primary reason for the absence of convenience API is that, unlike OpenSlide, Bio-Formats was not  designed with whole-slide imaging in mind. An additional complication is that it has support for many other modalities (high-content screening, fluorescence...) so we would need to think carefully for any domain-specific addition. Based on your Openslide experience, which kind of API would you consider added in Bio-Formats?

yaokkkkkk, 2021-06-16:
Maybe I don't have enough experience, the api to get the label image and the thumbnail image please

sbesson, 2021-06-21:
Thanks. Trying to clarify my question, we are aware of the [OpenSlide.get_thumbnail()](https://openslide.org/api/python/#openslide.OpenSlide.get_thumbnail) API. Is this the one  you have been using previously and for which you would like to have a Bio-Formats equivalent?
For the label image, which OpenSlide method would be used to retrieve this information?

closed 2023-03-06 by sbesson