ome/bioformats #3716: How to make OMERO server render a new image?

r0ach0, 2021-07-19:
I'm trying to add a new format reader to bioformats.
the reader works fine until images are uploaded to omero server.
omero.insight shows first and preview images.

but, in image viewer, any function effects on the image.
z and t dimension scrolls, channel on/off buttons, color range.

when i toggle channel button, the image turns black, and never comes back.
funny thing is, projection works fine through z and t dimensions.

i think openBytes() works pretty fine.
what do you think is the problem?

[image]

dgault, 2021-07-20:
Hi @r0ach0, is your new reader in its own jar or is it built with Bio-Formats? Have you included the reader with Insight and the OMERO Server?

r0ach0, 2021-07-21:
> Hi @r0ach0, is your new reader in its own jar or is it built with Bio-Formats? Have you included the reader with Insight and the OMERO Server?

Yes. I replaced bioformats-gpl-[VERSION].jar in Insight/app/lib, server/lib/client, server/lib/server.

closed 2021-07-26 by r0ach0

joshmoore, 2021-08-03:
@r0ach0 : can I ask that you please stop deleting the content of the questions you ask us? We're willing to help for free only because those questions & answers help others.