ome/bioformats #3722: ImageConverter not support openslide api

mywsat, 2021-08-20:
I use the bioformat API to convert JPG to tiff. It is found that the converted TIFF file (pyramid) cannot be opened with openslide. Only tile TIFF is supported
```java 
ImageConverter converter = new ImageConverter();
converter.in = src;
converter.out = out;
converter.bigtiff = true;
//converter.nobigtiff=false;
converter.overwrite = true;
converter.saveTileWidth = 1024;
converter.saveTileHeight = 1024;
converter.compression = "zlib";
//converter.merge=true;
converter.pyramidScale=2;
converter.pyramidResolutions=3;
// converter.flat=true;
converter.testConvert(new ImageWriter(), new String[]{});
```
only one level
[image]

sbesson, 2021-08-20:
Hi @mywsat:
-  the `TiffWriter` will be selected when passing an output filename with a default TIFF extension. The output format is a basic TIFF-based format (with some ImageJ metadata) and does not support pyramidal levels and will only write a single resolution for each plane
- the `OMETiffWriter` will be selected when passing an output filename with an OME-TIFF extension. The output format supports multi-resolution. However `openslide` has not been updated to read this open pyramidal format so it falls back to being detected as a [Generic tiled TIFF format](https://openslide.org/formats/generic-tiff/) and does not recognize the pyramidal levels.

If using `openslide` is part of your requirement, you will need to convert the data into a supported multi-resolution file format. In this case `libvips` might be an option to go from `jpg` to `Generic tiled TIFF` - see https://forum.image.sc/t/create-more-than-series-using-qupath/52849/9 for a typical workflow /cc @jcupitt

melissalinkert, 2023-09-12:
Closing in the absence of updates, since there are no planned changes here. @mywsat, if you're still having trouble and the suggestions in https://github.com/ome/bioformats/issues/3722#issuecomment-902668107 didn't help, feel free to re-open with more information about your workflow requirements.