ome/bioformats #3730: Custom reader supplied LUT gets lost when using ImageJ

ngladitz, 2021-09-23:
I am trying to evaluate the possibility of having the OBF reader supply custom LUTs via e.g. `get8BitLookupTable()`.
By default ("Color mode: Default") loading an OBF in ImageJ results in grayscale display.
Returning any LUT via `get8BitLookupTable()` seems to result (irregardless of actual content) in a gradient LUT display from black to yellow.

I found the following snippet:
https://github.com/ome/bioformats/blob/b67dfcac19b898e9a8e2481e5288bdc4b7aa782d/components/bio-formats-plugins/src/loci/plugins/in/ImagePlusReader.java#L193-L209

I don't understand it or know the history behind it or possible side effects but commenting it out seems to result in my custom LUT being used properly.

dgault, 2021-09-27:
Anything in bio-formats-plugins will be exclusively an issue for the ImageJ plugin so it looks likely that it is highlighting an issue with the plugin rather than the OBFReader. It might be worth testing your reader changes with the command line tools if that is easier for you than dealing with an issue in the plugin.

The section of the ImagePlusReader has been in the code base for quite a long time now (originally added in https://github.com/ome/bioformats/commit/49ea8371d2e9ffe9c28882811d43986e9ed21a6f#diff-f98a7d3f97d986d816cdeaf9a6544940c03bfb2f3a075cf4f523b89882a7e489) and it sounds like the original intention was to prevent custom LUT overriding the GrayScale option. The LUTs should still be preserved but Im guessing are not being applied properly further on in the stack.

Normally are you only seeing the black/yellow gradient in GrayScale mode or is it the same in Colorized or Composite?

ngladitz, 2021-09-28:
Thank you for looking at this.

Internally the wish came up that it would be nice to display OBFs in ImageJ / FIJI with the same or similar custom LUTs as they are in our own products (and if that works perhaps try collating related images into multi channel images).
Ideally the series selection dialog should also show thumnails colored by that custom LUT (currently they seem to be always gray).

So to start with I was only evaluating if this is something that I could do through the Bio-Formats reader.
There may be other consumers which use Bio-Formats outside of ImageJ and I obviously wouldn't mind if it were working there as well but for now that isn't part of the use case.

With ImagePlusReader unmodified I see:
	- Default: black/yellow
	- Composite: black/yellow
	- Colorized: black/yellow
	- Grayscale: grayscale
	- Custom: custom
	
With ImagePlusReader modified ("HACK" commented out) I see:
	- Default: my LUT as returned by the modified OBF reader
	- Composite: black/yellow
	- Colorized: black/yellow
	- Grayscale: my LUT
	- Custom: custom
	
So as you noted the hack does seem related to the grayscale setting and just commenting it does seem to break that.
I am unsure about the intended behavior of Composite and Colorized when a reader does supply a LUT; the custom LUT gets ignored even with the HACK commented out and I am unsure if / what other metadata is observed.

dgault, 2023-01-18:
A similar issue was reported on forum thread https://forum.image.sc/t/bio-formats-color-wrong-for-imagej-images/76021/7 and can reproduced with the sample file provided.

With that sample file, we have an OME-TIFF with a LUT and the reader is able to parse and read the LUT. The LUT is then read by the ImageProcessReader (https://github.com/ome/bioformats/blob/develop/components/bio-formats-plugins/src/loci/plugins/util/ImageProcessorReader.java#L289) and the LUT is then set on the processors (https://github.com/ome/bioformats/blob/develop/components/bio-formats-plugins/src/loci/plugins/util/ImageProcessorReader.java#L199-L222). As the file is using a ByteProcessor the LUT is saved to the processor as an IndexColorModel rather than a LUT (LUT extends IndexColorModel).

When the ImageProcessReader then checks the processor (https://github.com/ome/bioformats/blob/develop/components/bio-formats-plugins/src/loci/plugins/in/ImagePlusReader.java#L198) it fetches an IndexColorModel and not a LUT and therefore does not set the luts on the reader.

This then means that when the Colorizer attempts to fetch the LUT's from the ImagePlusReader (https://github.com/ome/bioformats/blob/develop/components/bio-formats-plugins/src/loci/plugins/in/Colorizer.java#L131) there are no properties set and when the Colorizer then attempts make the LUT's (https://github.com/ome/bioformats/blob/develop/components/bio-formats-plugins/src/loci/plugins/in/Colorizer.java#L131) it will instead apply defaults as none have been found.

A quick test to modify the ImagePlusReader to handle IndexColorModel as well as LUT seemed to work and apply the correct LUT in composite mode. A PR to properly implement this fix will follow.