ome/bioformats #3735: bfopen from the MATLAB toolbox does not read all z-frames in the .nd2 file

CreLox, 2021-09-30:
I am using MATLAB R2021a on my mac to open the attached example.nd2 _E. coli_ imaging file that has 15 z-frames (3 channels). However, `Results = bfopen('example.nd2')` gives me:

```
Results =

  1×4 cell array

    {42×2 cell}    {718 java.util.Hashtable}    {42×42 cell}    {1×1 loci.formats.ome.OMEXMLMetadataImpl}
```

The Bioformats_package.jar is up-to-date. By plotting all images, I found out that the 15th z-frame is not being read. My colleague using R2020b on his Windows can read all z-frames. I am not sure what is the issue here.
[example.nd2.zip](https://github.com/ome/bioformats/files/7261330/example.nd2.zip)

sbesson, 2021-10-04:
Thanks @CreLox, what you describe certainly reads confusing. What is the version of Bio-Formats in your MATLAB environment as displayed by

```
[~, version] = bfCheckJavaPath
```

Also have you tried opening the same sample file using  Bio-Formats command line utilities on your Mac and are all the Z sections correctly detected?

CreLox, 2021-10-04:
@sbesson Thanks for the reply. I have not tried the command line utilities but using Bio-Formats on ImageJ, I can read all 15 planes correctly.
`>> [~, version] = bfCheckJavaPath

version =

    '6.7.0'
`

sbesson, 2021-10-04:
Thanks, I retested on R2021a / OS X 11.6 using Bio-Foramts 6.7.0 and the sample file above and I definitely still get 45 individual planes.

So the dimensions are read incorrectly in your set up? What does the following code return?

```
>> r=bfGetReader('example.nd2');
>> r.getSizeT()
>> r.getSizeZ()
>> r.getSizeC()
```

CreLox, 2021-10-04:
Actually when I tried to read it again today it says,
```
Results =

  1×4 cell array

    {45×2 cell}    {1033 java.util.Hashtable}    {45×45 cell}    {1×1 loci.formats.ome.OMEPyramidStore}
```
Even I cannot repeat it anymore on my computer using the same computer/MATLAB distribution. I did not change any files/codes. But I am pretty sure the (x, y) dimensions were read correctly though, even in the previous situation. Because I did a lot of pixel counting per frame (512 x 512) in my application.
I noticed that Results{1, 4} has a different value than last time (when an incorrect number of z-frames was read). Could that be useful in figuring out the problem?

sbesson, 2021-10-06:
Thanks for letting us know. The fact you load the data correctly now is at least progress even if confusing.

The two differing elements contain the original metadata table and the object returned by `getMetadataStore()`. The fact the latter one has a different type makes me wonder whether a former version of Bio-Formats had not been used. Even if without changing any file, this could be caused by some modification/addition to the Java static or dynamic path that would conflict with the latest version of Bio-Formats.

Assuming you are reliably retrieving the correct dimensions, I am inclined to close this under the `wontfix` banner. If you get incorrect planes, I would say printing out the output of `loci.formats.FormatTools.VERSION` as well as the full Java path as part of the same session would be the first thing to do and we can follow-up from there.