ome/bioformats #3752: Errors when converting to tiled TIFF images with compression

petebankhead, 2021-12-10:
Congrats on the new release, it's great to see even more whole slide image support in v6.8.0!

While exploring, I think I found a potential regression for `ImageConverter` when converting TIFFs with compression.

I tested this on macOS with [CMU-1-Small-Region.svs](https://openslide.cs.cmu.edu/download/openslide-testdata/Aperio/), both using `ImageConverter` directly and via `bfconvert`.

Using
```
./bfconvert -overwrite -no-upgrade -series 0 -compression JPEG -tilex 256 -tiley 256  "/path/to/CMU-1-Small-Region.svs" "/path/to/converted.ome.tif"
```
takes ~4-5 seconds and results in an image as show below:

[image]

Switching compression to LZW gives a similarly corrupt image, and using JPEG-2000 gives an ome.tif that cannot be opened later.

Adding the new `-no-sequential` option appears to fixes all of these issues.

Alternatively, if I omit the tile size then I also get the correct output - but the conversion time jumps to 20-30 seconds (regardless of `-no-sequential`).

The conversion works in v6.7.0, requiring ~4-5 seconds regardless of whether tile size is specified or not.

dgault, 2021-12-10:
Hi @petebankhead, thank you for investigating and opening the issue. This unfortunately looks likely to be a regression caused by the introduction of the new `-no-sequential` option. We will investigate and aim to resolve this for a follow up release in the new year. Hopefully using 6.7.0 as a work around i the short term is suitable for now.

dgault, 2021-12-10:
As a follow up, the `-no-sequential` flag here is not the direct cause of the issue and is a bit of a red herring. The change which has produced the issue if bfconvert is https://github.com/ome/bioformats/pull/3709/files#diff-cb592158a6af58e0453459ed7811838aa3ec86efd6809c066020bd14b094c160L917 

By using the writer API for tiling we are no longer passing an IFD to saveBytes. This shouldn't be an issue as saveBytes will create an empty IFD if none is passed. However this seems to cause an issue with a combination of tiling and compression being used. Creating a single empty IFD outside of the tiling loop and passing it to saveBytes resolves the issue, but is a bit of a hack and should not be necessary. 

The reason the no-sequential flag does not throw errors is due to the conditional path at https://github.com/ome/bioformats/blob/develop/components/formats-bsd/src/loci/formats/tiff/TiffSaver.java#L462 which seeks to the next offset.

I believe what is happening here is that when using compression the bytesCounts per tile differ and when using a new IFD for each tile the resulting `IFD.TILE_BYTE_COUNTS` and `IFD.TILE_OFFSETS` end up incorrect.

Below are the correct offsets and byteCounts for a sample image when using saveBytes with the IFD API:
```
TIFF Directory at offset 0x8 (8)
  Image Width: 1834 Image Length: 1993
  Tile Width: 1024 Tile Length: 240
  Resolution: 21427, 21427 pixels/cm
  Bits/Sample: 8
  Sample Format: unsigned integer
  Compression Scheme: JPEG
  Photometric Interpretation: RGB color
  Samples/Pixel: 3
  Planar Configuration: separate image planes
  ImageDescription: ImageJ=
hyperstack=true
images=3
channels=3
slices=1
frames=1
  Software: OME Bio-Formats (unknown version)
  54 Tiles:
      0: [     768,    22087]
      1: [   60929,    18014]
      2: [  109635,    22048]
      3: [  169201,    18260]
      4: [  218837,    19269]
      5: [  272476,    18387]
      6: [  322592,    18681]
      7: [  375651,    15654]
      8: [  419033,    18754]
      9: [  472177,    13488]
     10: [  510058,    16951]
     11: [  558524,    10484]
     12: [  587792,    19615]
     13: [  643321,     9898]
     14: [  671339,    23810]
     15: [  736723,    16813]
     16: [  783448,    11402]
     17: [  815832,     9746]
     18: [   22855,    22666]
     19: [   78943,    18236]
     20: [  131683,    22418]
     21: [  187461,    18615]
     22: [  238106,    20565]
     23: [  290863,    18775]
     24: [  341273,    19757]
     25: [  391305,    16403]
     26: [  437787,    20159]
     27: [  485665,    13995]
     28: [  527009,    18183]
     29: [  569008,    10695]
     30: [  607407,    21312]
     31: [  653219,    10346]
     32: [  695149,    24609]
     33: [  753536,    17700]
     34: [  794850,    11561]
     35: [  825578,     9877]
     36: [   45521,    15408]
     37: [   97179,    12456]
     38: [  154101,    15100]
     39: [  206076,    12761]
     40: [  258671,    13805]
     41: [  309638,    12954]
     42: [  361030,    14621]
     43: [  407708,    11325]
     44: [  457946,    14231]
     45: [  499660,    10398]
     46: [  545192,    13332]
     47: [  579703,     8089]
     48: [  628719,    14602]
     49: [  663565,     7774]
     50: [  719758,    16965]
     51: [  771236,    12212]
     52: [  806411,     9421]
     53: [  835455,     8146]

```
And when using saveBytes without the IFD you get the incorrect values below:
```
TIFF Directory at offset 0x8 (8)
  Image Width: 1834 Image Length: 1993
  Tile Width: 1024 Tile Length: 240
  Resolution: 21427, 21427 pixels/cm
  Bits/Sample: 8
  Sample Format: unsigned integer
  Compression Scheme: JPEG
  Photometric Interpretation: RGB color
  Samples/Pixel: 3
  Planar Configuration: separate image planes
  ImageDescription: ImageJ=
hyperstack=true
images=3
channels=3
slices=1
frames=1
  Software: OME Bio-Formats (unknown version)
  54 Tiles:
      0: [     768,     9746]
      1: [   60929,     9746]
      2: [  109635,     9746]
      3: [  169201,     9746]
      4: [  218837,     9746]
      5: [  272476,     9746]
      6: [  322592,     9746]
      7: [  375651,     9746]
      8: [  419033,     9746]
      9: [  472177,     9746]
     10: [  510058,     9746]
     11: [  558524,     9746]
     12: [  587792,     9746]
     13: [  643321,     9746]
     14: [  671339,     9746]
     15: [  736723,     9746]
     16: [  783448,     9746]
     17: [  815832,     9746]
     18: [   22855,     9746]
     19: [   78943,     9746]
     20: [  131683,     9746]
     21: [  187461,     9746]
     22: [  238106,     9746]
     23: [  290863,     9746]
     24: [  341273,     9746]
     25: [  391305,     9746]
     26: [  437787,     9746]
     27: [  485665,     9746]
     28: [  527009,     9746]
     29: [  569008,     9746]
     30: [  607407,     9746]
     31: [  653219,     9746]
     32: [  695149,     9746]
     33: [  753536,     9746]
     34: [  794850,     9746]
     35: [  825578,     9877]
     36: [   45521,     9746]
     37: [   97179,     9746]
     38: [  154101,     9746]
     39: [  206076,     9746]
     40: [  258671,     9746]
     41: [  309638,     9746]
     42: [  361030,     9746]
     43: [  407708,     9746]
     44: [  457946,     9746]
     45: [  499660,     9746]
     46: [  545192,     9746]
     47: [  579703,     9746]
     48: [  628719,     9746]
     49: [  663565,     9746]
     50: [  719758,     9746]
     51: [  771236,     9746]
     52: [  806411,     9746]
     53: [  835455,     8146]
```

closed 2022-01-12 by dgault