ome/bioformats #3755: Drop loci_tools uber JAR

sbesson, 2021-12-13:
See https://github.com/ome/www.openmicroscopy.org/pull/555

This uber-JAR has been deprecated for almost a decade (superseded by `bioformats_package.jar`) and was solely maintained for backwards-compatibility.

Following Log4Shell (CVE-2021-44228), a security advisory was opened against this bundle - see [2021-SV4](https://www.openmicroscopy.org/security/advisories/2021-SV4/). The current workaround is to use the logback-based bioformats_package bundle.

For the next release of Bio-Formats, the proposal is to completely get rid of this bundle and the `log4j` dependency:

-  [ ] remove https://github.com/ome/bioformats/blob/develop/components/bundles/loci_tools
-  [ ] review remaining mentions of `log4j` in the code base
-  [ ] review documentation

ctrueden, 2021-12-13:
I looked for remaining references to `loci_tools.jar`. I fixed a couple of them (imagej/imagej.github.io@ad94a881b12ed549125b46254438ed271f600cb6, fiji/fiji@3428cdbb8fd6eaa6483ce1e084cc93108f6c8ca5, uw-loci/loci-scripts@bea4a2ede474eb481a35a2fbf7c18bba570980c1). Here are the remaining ones I found but did not fix:

## Vaa3D

https://github.com/Vaa3D/v3d_external
```
v3d_main/build.macx:#if [ -f bin/plugins/64bit/imageIO/load_image_using_Bioformats/loci_tools.jar ]; then
v3d_main/io/io_bioformats.cpp:    //look for loci_tools.jar
v3d_main/io/io_bioformats.cpp:        lociLibPath = getAppPath().append("/").append("loci_tools.jar");
v3d_main/io/io_bioformats.cpp:        lociLibPath = QFileDialog::getOpenFileName(0, QObject::tr("select the library of Bioformats Java library (loci_tools.jar)"),
v3d_main/io/io_bioformats.cpp:            v3d_msg("Cannot find loci_tools.jar, please download it and make sure it is put under the Vaa3D executable folder, parallel to the Vaa3D executable and the plugins folder.");
```

## Orbit

https://github.com/mstritt/orbit-image-analysis
```
src/main/java/com/actelion/research/orbit/imageAnalysis/utils/ImageProcessorReader.java:// Decompiled from loci_tools.jar (http://www.loci.wisc.edu/bio-formats/downloads)
src/main/java/com/actelion/research/orbit/imageAnalysis/utils/ImageProcessorReader.java:/* Location:           D:\dev\Java\OrbitImageAnalysis\lib\loci_tools.jar
```

## Microscopy Image Browser (MIB)

https://github.com/Ajaxels/MIB
```
ImportExportTools/BioFormats/bfopen3.m:%     to loci_tools.zip. If this happens, rename it back to loci_tools.jar.
ImportExportTools/BioFormats/selectLociSeries.m:%        (e.g., C:/Program Files/MATLAB/work/loci_tools.jar).
ImportExportTools/BioFormats/selectLociSeries.m:%     if ~isempty(strfind(javapath{i},'loci_tools.jar'))
ImportExportTools/BioFormats/selectLociSeries.m:%     javaaddpath(fullfile(fileparts(mfilename('fullpath')),'loci_tools.jar'));
```

## python-bioformats

https://github.com/CellProfiler/python-bioformats
```
bioformats/formatreader.py:    # This uses the reader.txt file from inside the loci_tools.jar
bioformats/formatwriter.py:    # This uses the writers.txt file from inside the loci_tools.jar
bioformats/metadatatools.py:        # Post loci_tools 4.2
bioformats/metadatatools.py:        # Post loci_tools 4.2 - use ome.xml.model.DimensionOrder
docs/index.rst:The javabridge package must be used to start the JVM with loci_tools.jar
setup.cfg:classpath = bioformats/jars/loci_tools.jar
```

jburel, 2021-12-14:
cc @mstritt

joshmoore, 2021-12-14:
see: https://github.com/CellProfiler/python-bioformats/issues/152

bethac07, 2021-12-14:
FWIW, it looks like since 2020 python-bioformats has been using `bioformats_package.jar` (so all versions of CellProfiler 4 should be good in that regard); it looks like other than setup.cfg (which apparently hasn't been touched in 8 years(!)), all of those are just comments which were not updated. I'll dig out those legacy comments.

https://github.com/CellProfiler/python-bioformats/commit/ad5df31cae90db5e053dd693b73979758052af7c#diff-f6e62418513760b4fb33dfae61cee65ed1e975085ed1a30f9945b1b30e76872a

joshmoore, 2021-12-14:
Thanks, @bethac07! There now definitely seems to be an aversion to anything that matches `git grep log4j` regardless of the impact.

closed 2022-01-11 by dgault