ome/bioformats #3756: Libjpeg-turbo exception when converting/opening .ndpi file.

michaellee1, 2021-12-15:
### Summary

I'm trying to convert an .ndpi file (Hamamatsu) to .tif, but the NDPI reader is throwing an error when I run conversion.

I've tried converting using:
1. bfconvert
2. python-bioformats bindings 

to read in the image, and I'm getting the exact same error on both:

### Exception
```
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'void org.libjpegturbo.turbojpeg.TJDecompressor.init()'
	at org.libjpegturbo.turbojpeg.TJDecompressor.init(Native Method)
	at org.libjpegturbo.turbojpeg.TJDecompressor.<init>(TJDecompressor.java:81)
	at loci.formats.services.JPEGTurboServiceImpl.getTile(JPEGTurboServiceImpl.java:346)
	at loci.formats.services.JPEGTurboServiceImpl.getTile(JPEGTurboServiceImpl.java:249)
	at loci.formats.in.NDPIReader.openBytes(NDPIReader.java:217)
	at loci.formats.FormatReader.openBytes(FormatReader.java:919)
	at loci.formats.ImageReader.openBytes(ImageReader.java:449)
	at loci.formats.tools.ImageConverter.getTile(ImageConverter.java:1085)
	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:798)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:727)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:1126)
```

I'm stuck here on what can be done to convert these images using bioformats - is there a way to resolve this issue?

petebankhead, 2021-12-17:
I am also unable to open .ndpi files because of libturbo issues if I am running on an ARM-based Mac *and* JDK intended for aarch64. However it works for me on the same Mac if I use a JDK intended for Intel processors.

Could this be the problem?

(Sorry for interjecting if not - I was meaning to report the Mac-specific behavior as it may soon become a blocker to releasing an apple-silicon-friendly version of QuPath.)

normanius, 2025-04-22:
I have the same issue... Has anyone been able to solve this problem in the meanwhile?

closed 2026-11-04 by sbesson