ome/bioformats #3761: ND2: Inverted LUT for RGB images

dgault, 2022-01-12:
Issue was raised on forum thread https://forum.image.sc/t/nd2-rgb-import-issue/60901 and a sample file reproducing the issue is available on Zenodo - https://zenodo.org/record/5772393#.YbNTrPnP1Zc

When the ND2 file is opened using Bio-Formats 6.8.0 and Composite mode via the FIJI plugin the resulting image does not match the colour seen when importing into NIS Elements Viewer

dgault, 2024-07-16:
An additional report of this issue has been received on forum thread https://forum.image.sc/t/nd2-rgb-import-issue/60901/7

A sample file and comparison tiff are available at https://zenodo.org/records/12734440

I was able to reproduce the same issue reported by the user using Bio-Formats 7.3.1 with the the red and blue channel appearing to be swapped when opening in composite mode with the Bio-Formats plugin in FIJI

melissalinkert, 2026-10-22:
See additional dataset in https://zenodo.org/records/23529989 (https://forum.image.sc/t/omero-display-rgb-nd2-images-not-in-the-correct-color/122877/3).