ome/bioformats #3774: Operetta reader improvement

NicoKiaru, 2022-02-10:
The operetta reader takes a lot of time to be initialized, typically tens of minutes for >k.100 Mb files.

It seems this part of the code is super slow:

https://github.com/dgault/bioformats/blob/989bd9f2f86a1494c12685d7f71a5ba3dbabf2e9/components/formats-gpl/src/loci/formats/in/OperettaReader.java#L387-L410

If I understand correctly, this loop is assigning each plane to its correct series and plane. It does this by looping through all planes for each row, col, field, z, c, t. This looks like a N^2 algorithm.

So assuming 10 rows, 10 cols, 4 fields, 5 z, 3 channels, 20 timepoints, you expect  120 000 planes, and because the loop goes through all indices, and all planes, we can expect 120 000 x 120 000 tests = 14.4 billions tests.

ping @sbesson @dgault

NicoKiaru, 2022-02-10:
Replacing with this:

```
        Map<String, Plane> hashToPlane =new HashMap<>();

        for (Plane p : planeList) {
            String key = p.row+":"+p.col+":"+p.field+":"+p.c+":"+p.z+":"+p.t;
            if (hashToPlane.containsKey(key)) {
                LOGGER.error("Multiple planes found for key {}",key);
            }
            hashToPlane.put(key, p);
        }

        for (int row=0; row<rows.length; row++) {
            for (int col=0; col<cols.length; col++) {
                for (int field=0; field<fields.length; field++) {
                    int nextPlane = 0;
                    for (int t=0; t<ts.length; t++) {
                        for (int z=0; z<zs.length; z++) {
                            for (int c=0; c<cs.length; c++) {
                                String key = rows[row]+":"+cols[col]+":"+fields[field]+":"+cs[c]+":"+zs[z]+":"+ts[t];
                                if (hashToPlane.containsKey(key)) {
                                    planes[nextSeries][nextPlane] = hashToPlane.get(key);
                                }
                                nextPlane++;
                            }
                        }
                    }
                    nextSeries++;
                }
            }
        }

```

brings a pretty significant speedup... from more than 20 minutes down to 30 seconds...

NicoKiaru, 2022-02-10:
See pr https://github.com/ome/bioformats/issues/3774

NicoKiaru, 2022-02-10:
See pr https://github.com/ome/bioformats/pull/3775

sbesson, 2022-03-15:
Closing as per #3775

closed 2022-03-15 by sbesson