ome/bioformats #3779: Another No Pixel Block error

EitNickS, 2022-02-21:
Hello,

Similar to the Case number #3242 and #3349, I am attempting to open up an OIR file on FIJI drag and drop bar --> Bio-Formats.

See file through drop box link (~125 MB): https://www.dropbox.com/s/9o7kob7ilkmwmqh/10kDa.P19.ZDC.7xConcentrationAlexProtocol.HomogenizedDex.oir?dl=0

When dragging any OIR file created after Feb 1st, I receive this:
Reading pixel blocks from .../10kda.P18.15xAlexSLOaLittleHigh.HomoDex.LowConfluency.oir
[WARN] No pixel blocks for plane #2
[WARN] No pixel blocks for plane #3
[WARN] No pixel blocks for plane #4
[WARN] No pixel blocks for plane #5
[WARN] No pixel blocks for plane #6
[WARN] No pixel blocks for plane #7
[WARN] No pixel blocks for plane #8
[WARN] No pixel blocks for plane #9
[WARN] No pixel blocks for plane #10
[WARN] No pixel blocks for plane #11
[WARN] No pixel blocks for plane #12
[WARN] No pixel blocks for plane #13

and I get multiple 0 pixel frames in the opened file through FIJI. 

I am a SUPER novice when it comes to FIJI, so please have some patience here, I tried to find a solution through the two case numbers stated above, but wasn't sure whether or not those were auto-applied to the most recent version of FIJI (as I have auto-updates on and just updated the FIJI version today) or if I have to download something specific (if so do please tell me plainly how to do that).

I do not have any additional plugins for Bio-Formats than what automatically comes with FIJI.

Also opening an old file doesn't show this warning. It makes me believe it was do to a new addition to our Olympus Microscope that is causing this issue (a zero drift compensator) and somehow that is changing meta data making FIJI/bio-formats struggle to read the file. When I converted the OIR to TIFF, I still had the same issue.

Please help, my research is at a standstill with this issue!!

Thanks,

Nick

dgault, 2022-02-22:
Hi @EitNickS, thank you for opening the issue and providing the sample files. In your FIJI can you check which version of Bio-Formats you currently have? You can find it by going to `Help > About Plugins > Bio-Formats Plugins`

EitNickS, 2022-02-22:
Absolutely! Bio-Formats release 6.8.1, build date jan 13, 2022.

dgault, 2022-02-23:
Thanks @EitNickS, I can confirm that I am seeing what looks like the same issue with the sample file with Bio-Formats 6.8.1. In my case the pixel block error is present and most of the frames load correctly, however the second Z slice is returning blank pixel frames which is incorrect. When opened with Olympus Viewer the frames are all present, so it looks as though this is indeed a bug in the Bio-Formats Reader. I will leave this Issue open for further investigation for a potential bug fix for this problem.

PR #4078 (2023-09-28): OIR: fix off-by-one in XML block reading
Fixes #3779.

Without this PR, `showinf` on the file in `curated/olympus-oir/gh-3779` should initialize the file and display images, but with several logged warnings about missing pixel data blocks as noted in the issue. The corresponding planes will be entirely blank (`-minmax -autoscale -fast` may help to verify).

With this PR, the same test should show no missing pixel data warnings, and all planes should contain reasonable data.

I wouldn't expect tests to fail, but it's probably also worth double-checking that build logs for `olympus-oir` do not c [... cut]
commit 68140426 (2023-09-28): OIR: fix off-by-one in XML block reading

Trimming the XML string before checking if it's valid meant that
cases where the first byte was a null byte were not being handled
correctly.

Fixes #3779.
closed 2023-09-28 by dgault