ome/bioformats #3781: Build Bioformats with Maven

FraLC42, 2022-02-23:
Hi all,
I just started using Maven and I am a total noob on coding, but I need bioformats tool to correct time in some lifetime fluorescence images in order to be able to use FLIMFit. 
I already installed Java and started Maven, but when i try to build Bioformats here is the error I get:
C:\Users\francesca.lacava>mvn 
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/C:/Windows/apache-maven-3.8.4/lib/bioformats_package.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/C:/Windows/apache-maven-3.8.4/lib/maven-slf4j-provider-3.8.4.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [ch.qos.logback.classic.util.ContextSelectorStaticBinder]
11:38:39.860 [main] DEBUG org.codehaus.plexus.PlexusContainer - Created new class realm maven.api
11:38:39.865 [main] DEBUG org.codehaus.plexus.PlexusContainer - Importing foreign packages into class realm maven.api
11:38:39.865 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: javax.annotation.* < plexus.core
11:38:39.866 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: javax.annotation.security.* < plexus.core
11:38:39.867 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: javax.inject.* < plexus.core
11:38:39.868 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.* < plexus.core
11:38:39.868 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.artifact < plexus.core
11:38:39.869 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.classrealm < plexus.core
11:38:39.869 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.cli < plexus.core
11:38:39.869 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.configuration < plexus.core
11:38:39.870 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.exception < plexus.core
11:38:39.870 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.execution < plexus.core
11:38:39.870 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.execution.scope < plexus.core
11:38:39.870 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.lifecycle < plexus.core
11:38:39.871 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.model < plexus.core
11:38:39.871 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.monitor < plexus.core
11:38:39.871 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.plugin < plexus.core
11:38:39.872 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.profiles < plexus.core
11:38:39.872 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.project < plexus.core
11:38:39.872 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.reporting < plexus.core
11:38:39.872 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.repository < plexus.core
11:38:39.873 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.rtinfo < plexus.core
11:38:39.873 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.settings < plexus.core
11:38:39.873 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.toolchain < plexus.core
11:38:39.874 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.usability < plexus.core
11:38:39.874 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.* < plexus.core
11:38:39.874 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.authentication < plexus.core
11:38:39.875 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.authorization < plexus.core
11:38:39.875 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.events < plexus.core
11:38:39.875 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.observers < plexus.core
11:38:39.876 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.proxy < plexus.core
11:38:39.876 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.repository < plexus.core
11:38:39.876 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.apache.maven.wagon.resource < plexus.core
11:38:39.877 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.classworlds < plexus.core
11:38:39.877 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.* < plexus.core
11:38:39.877 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.classworlds < plexus.core
11:38:39.878 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.component < plexus.core
11:38:39.878 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.configuration < plexus.core
11:38:39.878 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.container < plexus.core
11:38:39.878 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.context < plexus.core
11:38:39.879 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.lifecycle < plexus.core
11:38:39.879 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.logging < plexus.core
11:38:39.879 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.personality < plexus.core
11:38:39.879 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.util.xml.Xpp3Dom < plexus.core
11:38:39.880 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.util.xml.pull.XmlPullParser < plexus.core
11:38:39.880 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.util.xml.pull.XmlPullParserException < plexus.core
11:38:39.880 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.codehaus.plexus.util.xml.pull.XmlSerializer < plexus.core
11:38:39.881 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.* < plexus.core
11:38:39.881 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.artifact < plexus.core
11:38:39.881 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.collection < plexus.core
11:38:39.881 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.deployment < plexus.core
11:38:39.881 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.graph < plexus.core
11:38:39.882 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.impl < plexus.core
11:38:39.882 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.installation < plexus.core
11:38:39.882 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.internal.impl < plexus.core
11:38:39.882 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.metadata < plexus.core
11:38:39.883 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.repository < plexus.core
11:38:39.883 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.resolution < plexus.core
11:38:39.883 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.spi < plexus.core
11:38:39.884 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.transfer < plexus.core
11:38:39.884 [main] DEBUG org.codehaus.plexus.PlexusContainer -   Imported: org.eclipse.aether.version < plexus.core
[... 7 lines cut ...]
11:38:39.983 [main] DEBUG org.apache.maven.cli.MavenCli - Message styles: debug info warning error success failure strong mojo project
11:38:39.987 [main] DEBUG org.apache.maven.cli.configuration.SettingsXmlConfigurationProcessor - Reading global settings from C:\Windows\apache-maven-3.8.4\conf\settings.xml
[... cut ...]
11:38:40.128 [main] ERROR org.apache.maven.cli.MavenCli - For more information about the errors and possible solutions, please read the following articles: 
11:38:40.129 [main] ERROR org.apache.maven.cli.MavenCli - [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/NoGoalSpecifiedException

I am using Windows CMD (And i Could also use some help on how to actually run Bioformats line tools (tiffcomment and xml) because I did not quite actually understand what I should do
Sorry, I will learn

dgault, 2022-02-23:
Are the command line tools all that you need? If so you shouldn't need to worry about building with maven.

If you simply download the command line tools from https://www.openmicroscopy.org/bio-formats/downloads/
Unzip the tools and go to the unzipped folder in your command line.
From there you can run the commands that you wish (see https://docs.openmicroscopy.org/bio-formats/6.8.1/users/comlinetools/index.html)

FraLC42, 2022-02-23:
Thank you! Another thing: to modify the xml data from the image, do I have to load it from ImageJ?

dgault, 2022-02-23:
There are a few different ways you can get the XML to modify. I would recommend using tiffcomment:

- First output the existing XML: `tiffcomment path/to/myfile.ome.tiff > metadata.xml`
- Then modify the xml file that was generated
- Then set the modified XML: `tiffcomment -set 'metadata.xml' path/to/myfile.ome.tiff`

FraLC42, 2022-02-23:
Thank you, I did it!

closed 2022-02-25 by dgault