ome/bioformats #3786: Error using bfGetReader (line 93) Java exception occurred: loci.formats.UnknownFormatException: Unknown file format:

nessj216, 2022-02-27:
Hello,
I have installed the recent bfmatlab on my Mac. When I am using it in a Matlab image processing pipeline (https://github.com/GarciaLab/mRNADynamics) that uses bio-format to processes .nd2 files , I get the following error:
"Error using bfGetReader (line 93)
Java exception occurred:
loci.formats.UnknownFormatException: Unknown file format:
/Volumes/NESS_2/livemRNA/Data/RawDynamicsData/2018-10-23/Kr4_Kr4/._Acquisition_0.nd2
	at loci.formats.ImageReader.getReader(ImageReader.java:202)
	at loci.formats.ImageReader.setId(ImageReader.java:844)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelFiller.setId(ChannelFiller.java:234)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:293)

Error in bfopen (line 122)
r = bfGetReader(id, stitchFiles);

Error in ExportDataForFISH (line 1459)
        DSPINdata(iSeries,:) = bfopen([Folder, filesep,
        char(fileNameSort(iSeries))]);"

 I'm using Matlab R2017b, as this is recommended by the analysis pipeline. Let me know if you need more info.

dgault, 2022-02-28:
Hi @nessj216, thank you for reporting the issue. Are you able to provide a link to the sample file that we can test?
Are you seeing this problem with all of your ND2 files or just this one?

nessj216, 2022-02-28:
The script is processing the files and gets "stuck" on the first file. Here is the link to the first .nd2 data file https://drive.google.com/file/d/1mWXmVKeGAkInhBkrK4U7_FSI4G-yqVyl/view?usp=sharing

Alternatively, here is a link to the 6 files I am processing (just in case): https://drive.google.com/drive/folders/1yno66ctSSSPH2w6a4CdSvaXdTsViSG3B?usp=sharing

dgault, 2022-02-28:
Thanks @nessj216, I tested the sample files with MATLAB_R2017b and Bio-Formats 6.9.0 and was not able to reproduce the same error as the files could be read and opened as normal. When looking at the error you are seeing there appears to be an extra . in front of the filename: `/Volumes/NESS_2/livemRNA/Data/RawDynamicsData/2018-10-23/Kr4_Kr4/._Acquisition_0.nd2`

Is that correct? Does changing it to  `/Volumes/NESS_2/livemRNA/Data/RawDynamicsData/2018-10-23/Kr4_Kr4/_Acquisition_0.nd2` have any effect?

melissalinkert, 2023-08-11:
Closing since this wasn't reproduceable. @nessj216, if you still see a problem and following the suggestion from @dgault in https://github.com/ome/bioformats/issues/3786#issuecomment-1054653673 doesn't help, please feel to re-open this issue with more details.

closed 2023-08-11 by melissalinkert