ome/bioformats #3793: bfGetPlane

Samaneh1994, 2022-03-10:
How I can access whole planes in multi images (time frame) with this function?

dgault, 2022-03-10:
Hi @Samaneh1994, Im assuming this is in MATLAB?

If you haven't already seen it there are some instructions on reading and displaying data in https://docs.openmicroscopy.org/bio-formats/6.9.0/developers/matlab-dev.html#opening-an-image-file

Using the bfopen function is probably the most straightforward way to access the data, but to use bfGetPlane you would need to get the reader first and make sure to set the series first. The below is taken from https://docs.openmicroscopy.org/bio-formats/6.9.0/developers/matlab-dev.html#reading-from-an-image-file

```
reader = bfGetReader('path/to/data/file');
omeMeta = reader.getMetadataStore();
series1_plane1 = bfGetPlane(reader, 1);
```

To switch between series in a multi-image file, use the [setSeries(int)](https://downloads.openmicroscopy.org/bio-formats/6.9.0/api/loci/formats/IFormatReader.html#setSeries-int-) method. To retrieve a plane given a set of (z, c, t) coordinates, these coordinates must be linearized first using [getIndex(int, int, int)](https://downloads.openmicroscopy.org/bio-formats/6.9.0/api/loci/formats/IFormatReader.html#getIndex-int-int-int-)

```
% Read plane from series iSeries at Z, C, T coordinates (iZ, iC, iT)
% All indices are expected to be 1-based
reader.setSeries(iSeries - 1);
iPlane = reader.getIndex(iZ - 1, iC -1, iT - 1) + 1;
I = bfGetPlane(reader, iPlane);
```

Samaneh1994, 2022-03-10:
hi, @dgault  thanks for your complete answer. yes, I'm using MATLAB, I assume that I should use the last one for switching between series and develop it for accessing whole planes.

sbesson, 2022-03-15:
@Samaneh1994 closing this but let us know if there are still issues and we need to reopen it

closed 2022-03-15 by sbesson