ome/bioformats #3797: When opening the different resolution series from a BDV xml/hdf5 file, the properties has incorrect image dimensions.

odinsbane, 2022-03-17:
If I create an xml/hdf5 dataset then open the dataset with fiji the bio-formats plugin gets used. When there are multiple series for the different resolutions, all of the series have the same pixel dimensions in properties.

To reproduce.

1 open an image stack
2 save the file as bdv dataset.

    plugins -> bigdataviewer-> export current image as xml/hdf5

    Make sure there are multiple resolutions being exported.

3. Open the resulting .xml with Fiji and check the box for open all series.
4. Select each window, and check the image properties. They all have the same pixel dimensions.

I'm not sure if this is an issue for the bigdataviewer or an issue for bioformats. I can create an example if necessary.

Thank you

sbesson, 2022-03-18:
@odinsbane by pixel dimensions, do you refer to the XYZTC dimensions or the physical sizes? A public representative sample would definitely be handy to track down the issue if you can create a minimal one.

odinsbane, 2022-03-18:
Here is an image that I converted to xml/h5. When you open it there are two series, they're the same image at different resolutions. When I go to image -> properties it shows the pixel sizes as 0.173 um for both image stacks. 

https://www.dropbox.com/sh/ni1cvclnno71v9k/AAAcIhrlR7HOSFpSU-RBKGSCa?dl=0

melissalinkert, 2022-03-18:
I don't think we've ever really decided what the correct behavior for physical pixel sizes in flattened sub-resolutions should be. There is at least one other report of a similar issue (https://trello.com/c/INs2Dw2I/147-incorrect-physical-pixel-sizes-with-pyramids), and it wouldn't surprise me if other readers have this same problem.

Computing would be consistent with https://github.com/glencoesoftware/bioformats2raw/pull/134, but I'm not entirely convinced that the readers should be doing that.

sbesson, 2022-03-18:
:+1: let's discuss this on MOnday at the weekly @ome/formats meeting

closed 2026-09-22 by sbesson