ome/bioformats #3800: Fill color configuration

melissalinkert, 2022-03-21:
See #3795 and https://github.com/glencoesoftware/bioformats2raw/pull/65 for prior work on this topic.

Some formats allow sparse pixel data, which means that a particular image tile may contain pixels that have no value stored in the file(s). This is especially true for whole slide formats. Most readers fill these undefined pixels with 0s, which doesn't always match the vendor software or users' expectations. Brightfield slides in particular are usually expected to have undefined pixels set to 255 (i.e. white).

To better address this across all readers, one approach would be:

* Define a new `IFormatReader` method `Color getFillColor()` which returns the `ome.xml.model.primitives.Color` that will be used for undefined pixels (the return type is up for debate). By default this returns black/0.
* Define a new `IFormatReader` method `setFillColor(...)` which allows the color to be changed. This alters the value returned by `getFillColor()`.
* Add an option to bfconvert/bioformats2raw/etc. that calls `setFillColor`.
* Update to always use `getFillColor()` as the default pixel value. This might be worth adding to `FormatTools.checkPlaneParameters` (or some other method that all `openBytes` implementations call), and would need to be changed in the ~20 places where `Arrays.fill(buf, (byte) 0)` is called by `openBytes`.
* Update individual readers to override `getFillColor()` if color metadata is defined or can be inferred for a particular format. This could be limited to the readers for whole slide formats, at least to start.

Specific API changes are up for discussion, as is the priority since this would require at least a minor version update.

joshmoore, 2022-03-21:
In an ideal world, would this be a model update?

melissalinkert, 2022-03-21:
I think ideally, we would be able to set something like `NaN` as the default pixel value, and the fill color would be a true rendering setting that you could change in viewers and converters. No strong opinion on whether or not this belongs in the model right now though.

petebankhead, 2022-04-05:
FWIW this would be **enormously** helpful for some people using Bio-Formats via QuPath. Filling with zero has caused problems, e.g. https://forum.image.sc/t/qupath-0-2-thresholder-problem-zeiss-z1-wsi/41594

PR #3963 (2023-04-19): Add API to configure the value of undefined pixels
Fixes #3800.

This generalizes the work in https://github.com/glencoesoftware/bioformats2raw/pull/65 to allow a fill value to be set on all readers. Unless otherwise specified, the fill value will be 0, so the default behavior should be unchanged.

`ZeissCZIReader` is currently the only reader that overrides the default fill value. This is an extension of #3795.

`showinf` and `bfconvert` also have a new `-fill byte` option that uses the new API.

Any of the files in https://github.com/openmicroscopy/data_repo_config/pull/518 could be used to  [... cut]
commit 64e15966 (2023-04-19): Add API to configure the value of undefined pixels

Fixes #3800.

This generalizes the work in https://github.com/glencoesoftware/bioformats2raw/pull/65
to allow a fill value to be set on all readers. Unless otherwise specified, the fill value
will be 0, so the default behavior should be unchanged.

ZeissCZIReader is currently the only reader that overrides the default fill value.
This is an extension of #3795.

showinf and bfconvert also have a new `-fill byte` option that uses the new API.
closed 2023-04-19 by dgault