ome/bioformats #3816: VSI: Only part of image displayed

dgault, 2022-04-26:
Issue was raised on forum thread https://forum.image.sc/t/importing-a-vsi-composite-image-in-imagej-via-bioformats-plugin-how-do-i-get-a-high-resolution-macro-image/35282/6

When opening the file the first series looks to have the correct dimensions but the image displayed is incorrect. 
[image]

The next series only has a small sample of the image along with downsampled resolutions of this are and a macro image of the whole image.

A sample file showing the issue was provided via private link and reproduced the issue using Bio-Formats 6.9.1

melissalinkert, 2023-08-07:
Was the provided sample data uploaded to `inbox`? I was going to see if this issue is still relevant in 7.0.0, but could not find the referenced dataset.

melissalinkert, 2025-08-19:
It looks like no data was added to `inbox`, and I cannot see any links to relevant data in https://forum.image.sc/t/importing-a-vsi-composite-image-in-imagej-via-bioformats-plugin-how-do-i-get-a-high-resolution-macro-image/35282.

Without a dataset to reproduce this problem, there isn't much we can do. If anyone still experiences this issue with 8.3.0, please upload a complete dataset to https://zenodo.org/communities/bio-formats/records and we'll re-open.

closed 2025-08-19 by melissalinkert