ome/bioformats #3818: LOF file error; [Fatal Error] :1:1: Content is not allowed in prolog.

jil24, 2022-05-04:
I hope this is the correct place to report issues with the LOF reader; I know it's an external contribution.

I have an image that will not read and results in this error:
`
[Fatal Error] :1:1: Content is not allowed in prolog.
Content is not allowed in prolog.
org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
        at org.apache.xerces.parsers.DOMParser.parse(Unknown Source)
        at org.apache.xerces.jaxp.DocumentBuilderImpl.parse(Unknown Source)
        at loci.formats.in.LeicaMicrosystemsMetadata.LMSXmlDocument.initFromXmlString(LMSXmlDocument.java:109)
        at loci.formats.in.LeicaMicrosystemsMetadata.LMSXmlDocument.<init>(LMSXmlDocument.java:87)
        at loci.formats.in.LeicaMicrosystemsMetadata.LMSImageXmlDocument.<init>(LMSImageXmlDocument.java:37)
        at loci.formats.in.LeicaMicrosystemsMetadata.LofXmlDocument.<init>(LofXmlDocument.java:39)
        at loci.formats.in.LOFReader.isThisType(LOFReader.java:154)
        at loci.formats.FormatReader.isThisType(FormatReader.java:653)
        at loci.formats.ImageReader.isThisType(ImageReader.java:864)
        at loci.formats.ImageReader.getReader(ImageReader.java:193)
        at loci.formats.ImageReader.setId(ImageReader.java:844)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:523)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1141)
 *** One or more readers is misbehaving. See the debug output for more information. e.g.:
     loci.formats.in.LOFReader@52e6fdee -> java.lang.NullPointerException('null') ***
Exception in thread "main" loci.formats.UnknownFormatException: Unknown file format: C:\Users\lakej\Desktop\BMSEQIF-003-tonsil-high-c1-cd25positives.lof
        at loci.formats.ImageReader.getReader(ImageReader.java:202)
        at loci.formats.ImageReader.setId(ImageReader.java:844)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:523)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1141)
`

It's a 4-channel, 20-slice single position image.

I can share the image but it is quite large; I've 7-zipped it:
https://www.dropbox.com/s/s9w3kfpzrhpbz86/DAPI-CD4-CD45-CD25.7z?dl=0

jil24, 2022-05-06:
I was actually able to get the file to open by replacing the malformed part in a hex editor... LAS X does not seem to care about the error, though.

XLEFReaderForBioformats, 2022-05-09:
hi @jil24 , thank you for notifying us about this issue!
could you please specify how the LOF file was created? did you create it in LASX and then open it in FIJI right away or was the LOF file also manipulated by other software in between? could you also please tell us which LASX version you used?

jil24, 2022-05-09:
Software version is LAS X 3.7.4.23463. The file was saved using the auto save functionality in LAS X, then renamed within LAS X. otherwise it was not modified.

XLEFReaderForBioformats, 2023-01-02:
hi @jil24,
the fix for LASX side behavior will be available in LASX 3.8.0 (as well as other future LASX releases such as 4.6.0 and 6.2.0).
I adapted the bioformats side behavior only by adding a more verbose error messaging. Unfortunately, when reading of LOF XML fails, metadata needed for correct reading of image data is missing and images cannot be opened.

However, there are some workarounds possible:
- open project file (XLEF) that references this image --> in this case, LOF XML will be ignored and XLIF XML will be used instead
- open project in a current LASX version, copy image to new project, save this project and open the new LOF with bioformats --> LOF XML is created anew

jil24, 2023-01-03:
Glad to hear it. I am no longer working on this particular project, but I was able to complete it by changing the autosave format to TIF for the duration. I then extracted relevant metadata (stage coordinates) from the XLEF manually.