ome/bioformats #3823: Plugins: Add option for memoizer

dgault, 2022-05-12:
Porting this from the original Trello card: https://trello.com/c/lYaGdkhX/203-investigate-using-memoizer-with-plugins
Ideally this would include a configurable option to toggle the behaviour in the plugins settings

sbesson, 2022-05-12:
One of the primary questions might be to define where should the memo files be stored. Two options come to mind:

- either use the in-place caching functionality of the `Memoizer` i.e. write the memo file within the same directory as the file. The downside is that this might be limited by permissions issues e.g. when working with files in a read-only mount or in a folder without write access.
- or decide on the canonical ImageJ directory for storing Bio-Formats cache files similar to OMERO `BioFormatsCache`. There is some built-in ImageJ location for storing setings and preferences (https://imagej.nih.gov/ij/docs/guide/146-13.html) but I don't know if there is anything equivalent for cache.

Silver-Fang, 2022-05-14:
> One of the primary questions might be to define where should the memo files be stored. Two options come to mind:
>
> * either use the in-place caching functionality of the `Memoizer` i.e. write the [...]

My advice is to also provide an option for the user to decide where to store the memo files. It could also have a default value - the in-place path.

sbesson, 2022-05-16:
Thanks @Silver-Fang, I cannot speak for all use cases and workflows but conceptually enabling in-place memoization  by default and allowing to override the cache directory would work for me.

In terms of UI, we are already struggling with the fact that the `Bio-Formats Import Options` window is very dense:

[image]

Two possible options here might be:

- add a checkbox allowing to enable the memoization in the `Import Options` UI and allow to select a directory in the `Plugins Configuration`
[image]
- move both the memoization enabling  and the setting of the cache location to the `Plugins Configuration` window as advanced features

Implementation-wise, I suspect the way to go is to add another wrapper call in the `ImportProcess` similar to https://github.com/ome/bioformats/blob/2451425b38d9b7713adde0a0404c98ecf1ba1a4f/components/bio-formats-plugins/src/loci/plugins/in/ImportProcess.java#L538-L540.

joshmoore, 2022-05-17:
Mostly as a side note, I did wonder if this should go all the way to the IJ prefs to be persistent between application instances.