ome/bioformats #3824: bfconvert ignoring pattern file series specification

jil24, 2022-05-12:
With a pattern file `test.pattern`: `series_<0-1>.tif`
When i invoke `bfconvert test.pattern output.tif`

I get a single-page tiff as output when I would like multiple series
There is no error, the converter just seems to not increment the series and repeatedly reads the same series_0.tif. Only one series/page/image is written
Constructing the pattern like this `series_<0,1>.tif` has the same result

Specifying z-planes, channels, and timepoints in the pattern file works correctly.

Bioformats 6.9.1

dgault, 2022-05-13:
Hi @jil24, how are your files named, are the series marked as part of the filename?

jil24, 2022-05-13:
Yes. In the example they are named series_0.tif and series_1.tif

dgault, 2022-05-16:
Thanks, I have been able to reproduce the issue with the latest Bio-Formats 6.9.1, though it looks like this has been an issue going back quite a while (pre 5.5.1 at least). The patternFileReader is returning the correct list of used Files so it looks like this is a bug at the FileStitcher level. I will update this issue once I have confirmed the root cause, we should also be able to add a unit test to reproduce the bug.

dgault, 2022-05-18:
My initial attempt to resolve this was to expand out the series block as part of findPatterns so that the selected series would get added and treated as external series in FileStitcher at https://github.com/ome/bioformats/blob/develop/components/formats-bsd/src/loci/formats/FileStitcher.java#L977

However there are also updates required throughtout the FileStitcher API anywhere get/setSeries and resolution are being used. I will try to continue working on this.