ome/bioformats #3825: Fiji opening .ims files taking unreasonably long time

Marien-kaefer, 2022-05-16:
Dear BioFormats Team,

I am regularly opening .ims files (from Dragonfly microscope) in Fiji. At the moment, the opening of large files takes a very long time, e.g. 32 mins for a 3.16 GB file at the maximum resolution level. I am sure that it did not take this long in the past and the RAM of the machine I am using is more than capable to host this amount of data. Saving the file as TIF (5.95GB) in the same location and re-opening it is much quicker, takes just over a minute.

When opening the .ims file as a virtual stack, it takes a long time for a new z-slice to load or.

I am using Fiji/ImageJ 1.53q, Java 1.8.0_172, BioFormats version 6.9.1
I downloaded Fiji and tried again - same problem.

I find that the workstations in the facility are having the same problem. Has anybody else got the same problem on their end?

Thanks,

Marie

dgault, 2022-05-16:
Cross linking to the image-sc post: https://forum.image.sc/t/fiji-opening-ims-files-taking-unreasonably-long-time/66986

dgault, 2022-05-17:
Thank you @Marien-kaefer for providing the sample file for testing. Having profiled the reading of a full series it appears that this is due to the compression used to store the data. When opening each chunk of data it first has to be decompressed and this is why the converted Tiff file reads much quicker but has a greater file size. The time spent is almost entirely in the underlying library used to read HDF files, and over 77% of the time is directly spent on decompressing/inflating (the call yucar.nc2.iosp.hdf5.H5tiledLayoutBB$DataChunk.inflate0 below). If the greater file size is not an issue for you I would suggest saving the files without the compression or converting to uncompressed data as you did with the Tiff.

[image]

Marien-kaefer, 2022-05-17:
Hi @dgault, 
Thank you so much for looking into the issue. The files come in .ims format straight from the microscope. I will raise the issue with the supplier (Andor) and/or Imaris directly. 
Thanks for looking into this.

dgault, 2022-05-18:
No problem, I will close this issue as I don't think there is much else we can do from the Bio-Formats side for now

closed 2022-05-18 by dgault