ome/bioformats #3846: [RFE] - New Zeiss CZI DetectorType

jdunhamvrtx, 2022-07-08:
Hi @BioFormats, 

Just a small enhancement request. Could you add GaAsP-PMT' to the DetectorType list? The relevent error is below.

WARN   l.enums.handlers.DetectorTypeEnumHandler - Unknown DetectorType value 'GaAsP-PMT' will be stored as "Other"

-J

dgault, 2022-07-08:
Hi @jdunhamvrtx, thank you for raising the issue, I have tagged this as an enhancement request. For now the warning message itself can be ignored, while the OME-XML will have the Detector listed as Other, the original value of GaAsP-PMT should remain present in the original metadata and can still be accessed if required.

melissalinkert, 2022-07-14:
Using enums for most of the `Type` values (`Detector`, `Microscope`, etc.), `Correction` and `Immersion` on `Objective`, and other instrument/experiment metadata is a design decision that dates back to the original 2003 OME schema. The intention was to standardize around a limited and predictable vocabulary. While we've discussed relaxing certain enums in the past (https://trello.com/c/zKLdSC6n/114-enumeration-flexibility), we're unlikely to switch to using free-form text for these values in the foreseeable future. At this point, that's a major schema change with impact across the whole OME ecosystem.

Adding values to the existing enums is a relatively minor change that we could include with the next set of schema changes. We have other candidates for enum expansion from the past few years, in particular https://trello.com/c/FxSUHfGp/273-add-more-model-enum-values and https://trello.com/c/Ds0zwZcF/458-add-airyscan-detector-type.

Moving this issue to https://github.com/ome/ome-model, since that's where schema work happens. Bio-Formats just uses the enums defined by the schema, so this isn't a something that would be changed in Bio-Formats directly.