ome/bioformats #3858: UnsatisfiedLinkError for ome.jxrlib when opening a .czi image

JKelle, 2022-08-09:
I get the following `java.lang.UnsatisfiedLinkError` when I try to open a .czi image:

```
11:51:31.270 [Preview] DEBUG loci.common.NIOByteBufferProvider - Using mapped byte buffer? false
11:51:31.282 [Preview] INFO loci.formats.ImageReader - ZeissCZIReader initializing test.czi
11:51:31.283 [Preview] DEBUG loci.formats.FormatHandler - ZeissCZIReader initializing test.czi
11:51:31.283 [Preview] DEBUG loci.formats.FormatHandler - loci.formats.in.ZeissCZIReader.initFile(test.czi)
11:51:31.381 [Preview] DEBUG loci.common.Location - Location.mapFile: image.czi -> loci.common.ByteArrayHandle@1f579ffc
11:51:31.381 [Preview] DEBUG loci.formats.FormatHandler - ZeissCZIReader initializing image.czi
11:51:31.381 [Preview] DEBUG loci.formats.FormatHandler - loci.formats.in.ZeissCZIReader.initFile(image.czi)
11:51:31.426 [Preview] DEBUG loci.formats.FormatHandler - Expected positive value for PhysicalSize; got 0.0
11:51:31.426 [Preview] DEBUG loci.formats.FormatHandler - Expected positive value for PhysicalSize; got 0.0
11:51:31.457 [Preview] DEBUG loci.common.Location - Location.mapFile: image.czi -> null
11:51:31.480 [Preview] DEBUG loci.common.Location - Location.mapFile: image.czi -> loci.common.ByteArrayHandle@3b91744f
11:51:31.480 [Preview] DEBUG loci.formats.FormatHandler - ZeissCZIReader initializing image.czi
11:51:31.480 [Preview] DEBUG loci.formats.FormatHandler - loci.formats.in.ZeissCZIReader.initFile(image.czi)
11:51:31.536 [Preview] DEBUG loci.common.Location - Location.mapFile: image.czi -> null
11:51:31.681 [Preview] DEBUG org.scijava.nativelib.NativeLibraryUtil - processor is UNKNOWN os.arch is aarch64
11:51:31.681 [Preview] DEBUG org.scijava.nativelib.NativeLibraryUtil - architecture is UNKNOWN os.name is mac os x
11:51:31.681 [Preview] WARN org.scijava.nativelib.NativeLibraryUtil - No native library available for this platform.
Exception in thread "Preview" java.lang.UnsatisfiedLinkError: 'long ome.jxrlib.JXRJNI.new_DecodeContext()'
	at ome.jxrlib.JXRJNI.new_DecodeContext(Native Method)
	at ome.jxrlib.DecodeContext.<init>(DecodeContext.java:43)
	at ome.jxrlib.AbstractDecode.decodeFirstFrame(AbstractDecode.java:178)
	at ome.jxrlib.Decode.decodeFirstFrame(Decode.java:72)
	at loci.formats.services.JPEGXRServiceImpl.decompress(JPEGXRServiceImpl.java:75)
	at loci.formats.codec.JPEGXRCodec.decompress(JPEGXRCodec.java:94)
	at loci.formats.in.ZeissCZIReader$SubBlock.readPixelData(ZeissCZIReader.java:4047)
	at loci.formats.in.ZeissCZIReader$SubBlock.readPixelData(ZeissCZIReader.java:4003)
	at loci.formats.in.ZeissCZIReader$SubBlock.readPixelData(ZeissCZIReader.java:3998)
	at loci.formats.in.ZeissCZIReader.openBytes(ZeissCZIReader.java:414)
	at loci.formats.FormatReader.openBytes(FormatReader.java:919)
	at loci.formats.FormatReader.openBytes(FormatReader.java:890)
	at loci.formats.FormatTools.openThumbBytes(FormatTools.java:1308)
	at loci.formats.FormatReader.openThumbBytes(FormatReader.java:940)
	at loci.formats.ImageReader.openThumbBytes(ImageReader.java:479)
	at loci.formats.ReaderWrapper.openThumbBytes(ReaderWrapper.java:360)
	at loci.formats.gui.BufferedImageReader.openThumbImage(BufferedImageReader.java:100)
	at loci.formats.gui.PreviewPane.run(PreviewPane.java:285)
	at java.base/java.lang.Thread.run(Thread.java:829)
```

I'm running from my M1 Mac, using macOS Monterey version 12.3.1.
I'm running the latest (6.10.0) `bioformats_package.jar` which I downloaded from https://downloads.openmicroscopy.org/bio-formats/6.10.0/artifacts/ .
I'm running the jar via Terminal with `java -jar bioformats_package.jar`.
I also tried to open this .czi image in an ubuntu:20.04 arm64 docker container, and I got the same Exception (though in this case I ran bioformats via python-bioformats instead of running the jar directly).

I can open other .czi files, just not this one. I'm not sure why this one is different, but based on the error message I'm guessing it's compressed with JPEG-XR format.
And I'm also inferring that `bioformats_package.jar` didn't install the ome.jxrlib dependency? Or is that something I need to install separately?

I have Fiji installed with the Bioformats plugin, and it can open this .czi file just fine.
I explicitly used the Bioformats importer (Plugins > Bio-Formats > Bio-Formats Importer), not the default (File > Open) file opening functionality.
After looking around Fiji's directory, I see the following jars:
    Fiji.app/plugins/bio-formats_plugins-6.9.0.jar
    Fiji.app/jars/bio-formats/formats-api-6.9.0.jar
    Fiji.app/jars/bio-formats/formats-bsd-6.9.0.jar
    Fiji.app/jars/bio-formats/formats-gpl-6.9.0.jar
    Fiji.app/jars/bio-formats/jai_imageio-0.0.0-STUB.jar
    Fiji.app/jars/bio-formats/jxrlib-all-0.2.4.jar
    Fiji.app/jars/bio-formats/metakit-5.3.3.jar
    Fiji.app/jars/bio-formats/ome-codecs-0.3.1.jar
    Fiji.app/jars/bio-formats/ome-common-6.0.7.jar
    Fiji.app/jars/bio-formats/ome-jai-0.1.0.jar
    Fiji.app/jars/bio-formats/ome-mdbtools-5.3.2.jar
    Fiji.app/jars/bio-formats/ome-poi-5.3.4.jar
    Fiji.app/jars/bio-formats/ome-xml-6.2.3.jar
    Fiji.app/jars/bio-formats/specification-6.2.3.jar
    Fiji.app/jars/bio-formats/turbojpeg-6.9.0.jar
These jars align with those listed in the documentation for changing the bioformats version for Fiji: https://docs.openmicroscopy.org/bio-formats/6.10.0/users/fiji/index.html .
However, that doc says to download these jars from http://downloads.openmicroscopy.org/bio-formats/ but I don't see these jars for any recent versions. Where can I get these jars from?

Questions
1. How can I get bioformats to open my .czi image?
2. Are recent versions of `bioformats_package.jar` missing the ome.jxrlib dependency? And/or do I need to install this separately?
3. Where are Fiji's bioformats jars coming from? Should I use those jars instead of `bioformats_package.jar`?

dgault, 2022-08-10:
Hi @JKelle, thank you for opening the issue.  The bioformats_package.jar should contain all the required dependencies, and having checked the 6.10.0 version at the download link the jxr dependencies are included as expected.

It very much looks as though the problem is related to your 4th question, the jxrlib dependency (which is included in the package jar) does not appear to have located a native library for the M1 arm architecture, we would likely need arm64 builds of the jxrlib to get this working. 

What is interesting is that your FIJI install can open the same JPEG-XR file. Have you installed any additional plugins or is it a base FIJI install?

melissalinkert, 2022-08-10:
I can reproduce this problem on an M1 Mac.

.czi files with JPEG-XR compression require the jxrlib library to decompress image data. As @dgault says, jxrlib is included in bioformats_package.jar, but does not yet include M1 support. I have opened an issue to fix this on the jxrlib repository: https://github.com/glencoesoftware/jxrlib/issues/30

Most Fiji installs include their own JRE, which for Macs is typically x86_64 architecture. You can check this by clicking `Plugins > Utilities > ImageJ Properties` and looking for the `os.arch` line in the window that pops up. jxrlib does include support for x86_64. At the moment, using an x86_64 JRE or JDK is the only workaround for this problem.

JKelle, 2022-08-12:
Thank you @melissalinkert for confirming it's an architecture issue, and for opening an issue with jxrlib!
Once jxrlib adds an M1 build, will that be included in an updated version of bioformats_package.jar?

dgault, 2022-08-17:
Yes, once there is a jxrlib build for M1 we will ensure to have a follow up Bio-Formats release with the included library.