ome/bioformats #3866: Leica LIF: Incorrect reading of data with FLIM and TileScan

dgault, 2022-08-17:
This issue was raised in forum thread https://forum.image.sc/t/data-import-bug-from-leica-containing-flim-and-tilescan-from-navigator/23551

Migrating this Issue from the original Trello card: 

A sample dataset is available for download and the issues have been reproduced with this dataset and Bio-Formats 6.0.0

When opening a Leica LIF-Project, which contains an Tilescan and at least two images with FLIM data, from the second one, data is not read correctly. Images appear skewed as if the dimensions or pixel size are incorrect.

Projects with Tilescan and data without FLIM work fine.

When Cropping the Tilescan (inside Leica software) data without FLIM can be read although there are some error messages (FIJI errors).
When Cropping the Tilescan (inside Leica software) data with FLIM causes and Exception (Bio-Formats IndexOutOfBounds exception).

Additionally a large amount of the metadata appears to be either incorrect or missing entirely, both from the OME metadata and the global metadata, this is particularly noticeable for channel metadata.

@XLEFReaderForBioformats, do you think you would be able to help with improving the metadata parsing for FLIM datasets?

XLEFReaderForBioformats, 2022-08-22:
Hi @dgault, 

thank you for raising this issue. We need to do some internal investigations on this topic, I'll let you know as soon as we know more!

Best
Constanze

dgault, 2022-09-12:
Hi @XLEFReaderForBioformats, Im reviewing the release milestones for the coming weeks and months and though I would follow up.  Do you think it is likely to expect an update that could be considered for on of the upcoming milestones (https://github.com/ome/bioformats/milestones)? We have some flexibility with the timelines if that would be helpful

XLEFReaderForBioformats, 2022-10-04:
Hi @dgault!
sorry for the late answer, i was on vacation in between... we discussed the release planning and decided that we can support 6.12.0 earliest, since i currently have a lot of items for our own releases on my desk... do you already have a rough idea about the release time span of 6.12.0?

dgault, 2022-10-04:
Thats no problem, we don't have a release timetable for 6.12.0 as of yet and have some flexibility around the schedule, I suspect it would likely be early in the new year at this point. If you have a timeframe you are targeting just let us know and we can try align those with a release.

XLEFReaderForBioformats, 2022-10-05:
Hi @dgault , thanks for the info!
we will be able to plan the timeframe and the list of supported items in November, after our upcoming releases. Then I'll come back to you.

XLEFReaderForBioformats, 2022-12-13:
Hi @dgault ,
sorry for the late update... I will be able to support bioformats in January and ~1st week of February. I'm planning to address all the bugs that I found listed here for LIF and LOF reader (and for XLEF reader as well if bugs also apply for it).

imagejan, 2023-08-03:
We're having the same issue and see wrong pixel data offsets in later series of a multiseries lif file.

@dgault I can provide an example dataset (.lif and .lifext if necessary) to reproduce the issue. Please let me know if and where I should upload it.
(I quickly checked http://qa.openmicroscopy.org.uk/qa/upload/ but it requires Flash Player which I don't have installed and isn't supported anymore by Adobe.)

dgault, 2023-08-03:
Hi @imagejan, yeah the old QA upload is no longer supported. If you were able to upload the dataset to https://zenodo.org/ that would be perfect

imagejan, 2023-08-16:
@dgault sorry for the delay. Here's a link to the example dataset on zenodo: https://doi.org/10.5281/zenodo.8252368