ome/bioformats #3879: LIFReader: java.lang.IllegalArgumentException when calling openBytes on some multi-series lifext samples

sbesson, 2022-09-21:
Using the [public samples](https://downloads.openmicroscopy.org/images/Leica-XLEF) uploaded as part of the work on #3747, an error occurs when reading the binary data of the last series of the `lif/dimension tests LIFs/XYZCS.lifext` sample file (from series 16

```
showinf /uod/idr/repos/curated/leica-xlef-lof/public/lif/dimension\ tests\ LIFs/XYZCS.lifext -series 16
Checking file format [Leica Image File Format]
Initializing reader
LIFReader initializing /uod/idr/repos/curated/leica-xlef-lof/public/lif/dimension tests LIFs/XYZCS.lifext
Reading header
Finding image offsets
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Initialization took 0.22s

Java does not support display of unnormalized floating point data.
Please use the '-normalize' option to avoid receiving a cryptic exception.

Reading core metadata
filename = /uod/idr/repos/curated/leica-xlef-lof/public/lif/dimension tests LIFs/XYZCS.lifext
Series count = 28
Series #0 :

...
Series #27 :
	Image count = 2
	RGB = false (1) 
	Interleaved = false
	Indexed = true (false color)
	Width = 256
	Height = 1
	SizeZ = 1
	SizeT = 1
	SizeC = 2
	Tile size = 256 x 1
	Thumbnail size = 128 x 1
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = float
	Valid bits per pixel = 32
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0

Reading series #16 pixel data (0-9)
Exception in thread "main" java.lang.IllegalArgumentException
	at java.nio.Buffer.position(Buffer.java:244)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:650)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:494)
	at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:675)
	at loci.formats.in.LIFReader.openBytes(LIFReader.java:358)
	at loci.formats.FormatReader.openBytes(FormatReader.java:919)
	at loci.formats.ImageReader.openBytes(ImageReader.java:449)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.gui.BufferedImageReader.openImage(BufferedImageReader.java:86)
	at loci.formats.tools.ImageInfo.readPixels(ImageInfo.java:821)
	at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1055)
	at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)
```

The issue only affects the `lifext` file.  The other `lif` files in this dataset  are detected as multi-images which read without issue.

dgault, 2022-12-01:
Another report of this issue was submitted via QA-32456, though it did not contain a sample file.

dgault, 2023-02-07:
Two further reports of this issue have been received in `Insight Bug-32574` and `Insight Bug-32576`. There were no sample files included with the bug reports. 

The stack traces included with the reports differ from the one posted previously in this issue:

```
Parameters: {stacktrace=java.lang.IllegalArgumentException: newPosition > limit: (664733 > 664732)
	at java.base/java.nio.Buffer.createPositionException(Buffer.java:318)
	at java.base/java.nio.Buffer.position(Buffer.java:293)
	at java.base/java.nio.ByteBuffer.position(ByteBuffer.java:1094)
	at java.base/java.nio.ByteBuffer.position(ByteBuffer.java:262)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:650)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:494)
	at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:675)
	at loci.formats.in.LIFReader.openBytes(LIFReader.java:358)
	at loci.formats.ImageReader.openBytes(ImageReader.java:465)
	at loci.formats.ChannelFiller.openBytes(ChannelFiller.java:167)
	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:229)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.MinMaxCalculator.openBytes(MinMaxCalculator.java:269)
	at ome.services.blitz.repo.ManagedImportRequestI.parseDataByPlane(ManagedImportRequestI.java:872)
	at ome.services.blitz.repo.ManagedImportRequestI.parseData(ManagedImportRequestI.java:803)
	at ome.services.blitz.repo.ManagedImportRequestI.pixelData(ManagedImportRequestI.java:676)
	at ome.services.blitz.repo.ManagedImportRequestI.step(ManagedImportRequestI.java:525)
	at omero.cmd.HandleI.steps(HandleI.java:448)
	at omero.cmd.HandleI$RunSteps.innerWork(HandleI.java:509)
	at omero.cmd.HandleI$2.doWork(HandleI.java:383)
	at omero.cmd.HandleI$2.doWork(HandleI.java:380)
	at jdk.internal.reflect.GeneratedMethodAccessor294.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.springframework.aop.support.AopUtils.invokeJoinpointUsingReflection(AopUtils.java:333)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.invokeJoinpoint(ReflectiveMethodInvocation.java:190)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:157)
	at ome.services.util.Executor$Impl$Interceptor.invoke(Executor.java:568)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:179)
	at ome.security.basic.EventHandler.invoke(EventHandler.java:154)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:179)
	at org.springframework.orm.hibernate3.HibernateInterceptor.invoke(HibernateInterceptor.java:119)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:179)
	at org.springframework.transaction.interceptor.TransactionInterceptor$1.proceedWithInvocation(TransactionInterceptor.java:99)
	at org.springframework.transaction.interceptor.TransactionAspectSupport.invokeWithinTransaction(TransactionAspectSupport.java:282)
	at org.springframework.transaction.interceptor.TransactionInterceptor.invoke(TransactionInterceptor.java:96)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:179)
	at ome.tools.hibernate.ProxyCleanupFilter$Interceptor.invoke(ProxyCleanupFilter.java:249)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:179)
	at ome.services.util.ServiceHandler.invoke(ServiceHandler.java:121)
	at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:179)
	at org.springframework.aop.framework.JdkDynamicAopProxy.invoke(JdkDynamicAopProxy.java:213)
	at com.sun.proxy.$Proxy81.doWork(Unknown Source)
	at ome.services.util.Executor$Impl.execute(Executor.java:447)
	at omero.cmd.HandleI.run(HandleI.java:379)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at ome.services.util.Executor$Impl$1.call(Executor.java:488)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)
, message=newPosition > limit: (664733 > 664732)}
```