ome/bioformats #3903: Virus Alert in formats-bsd.jar (version 6.10.0 and higher)

bvandenbroek, 2022-11-04:
Hello,

Our institute uses Fortinet antivirus. Since a few days I am getting a virus alert in `formats-bsd.jar` (shipped/updated with Fiji).
The alert (Java/Agent.ES!tr) results in Fortinet blocking access to the file. Consequently, running the Fiji Updater causes a runtime error that makes any further/other updates impossible.

@ctrueden and @axtimwalde have greatly helped in identifying the issue. It appears that Fortinet sounds the alarm bells for `formats-bsd.jar` version `6.10.0` and higher. Why the issue surfaces only now probably has to do with the version of the antivirus engine version: on computers with an AV build version lower than 00266 nothing is detected.

Now, upon *uploading* the files (rather than sending the URL) to VirusTotal  [a threat is detected](https://www.virustotal.com/gui/file/9157394b38c2c3524bb512c77913ecefa66a7eff44cf955750e76013f9bc1c92/detection) in [formats-bsd-6.10.1.jar](https://artifacts.openmicroscopy.org/artifactory/ome.releases/ome/formats-bsd/6.10.0/formats-bsd-6.10.0.jar), but [formats-bsd-6.9.1.jar](https://artifacts.openmicroscopy.org/artifactory/ome.releases/ome/formats-bsd/6.9.1/formats-bsd-6.9.1.jar) appears to be [clean](https://www.virustotal.com/gui/file/e9e697550c86608e9449aaa46cfa60ddef464ddfa637cdeba84bd1ca0c019463).

It seems unlikely that there really is malicious code present in the `.jar`, and possibly the fault is entirely on Fortinet's side. That said, could anyone please have a look at it soon, and perhaps provide an updated version that doesn't trigger the alarm, because I'm expecting a large number of complaints when our institute further rolls out antivirus engine updates.

Thanks,
Bram

ctrueden, 2022-11-05:
Some discussion took place [in the fiji/fiji channel on Gitter](https://gitter.im/fiji/fiji?at=6363d420f00b697fec70b2e1). @axtimwalde suggested to rebuild these versions of formats-bsd from source on a fresh computer at the same released versions, and then compare the resulting contents vs the originally flagged binaries. There may be differences in timestamps in the `META-INF/MANIFEST.MF` and such, but otherwise everything should be identical if built with the same version of Java, right?

bvandenbroek, 2022-11-05:
Not sure if it helps in narrowing down, but my local Fortinet virus alert says that the problem is located in `SlideBook7Reader$CSBFile70$1$1.class`

dgault, 2022-11-07:
Thanks @bvandenbroek and @ctrueden for raising the issue and for your initial investigation. This is the first time we have seen these reports and we will be investigating this internally as a priority. So far this appears to be only Fortinet  that is flagging the jar. 

My initial reading of it is that the detection started with the introduction of the `Slidebook7Reader` which included a new `snakeyaml` dependency (https://github.com/ome/bioformats/pull/3731/files). The inclusion of the `snakeyaml` dependency I believe is likely being flagged as a vulnerability for https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2022-38752. It looks like this may have been resolved in a later version 1.3.2 onwards, however when testing if an upgrade to the dependency would resolve the issue I am still seeing the same report for Fortinet.

We will continue investigating on our side and may also look at raising this with Fortinet via their classification dispute form https://www.fortiguard.com/faq/classificationdispute

PR #3906 (2022-11-21): Slidebook7: Remove nested inner class
This is an attempt to fix the issue raised in https://github.com/ome/bioformats/issues/3903

From the Fortinet error report it shows the error within `SlideBook7Reader$CSBFile70$1$1.class` which suggests that the issue was within the second anonymous  inner class within CSBFile70. The second nested `FileFilter` looks to be easily removed and results in the jar passing all of the security checks.

To test:
- Ensure builds and tests continue to pass
- Without this PR the compiled `formats-bsd` jar should fail the Fortinet check in https://www.virustotal.com/ [... cut]
closed 2022-11-21 by dgault