ome/bioformats #3914: bfconvert: IndexOutOfBoundsException when saving bytes

dgault, 2022-11-28:
This issue was raised on forum thread https://forum.image.sc/t/ome-tif-multi-channel-tiled-pyramid-from-individual-large-tiff-raw-files/23690/25

A sample file has been provided that reproduces the issue with Bio-Formats 6.11.0, it can be found here: https://doi.org/10.5281/zenodo.7337371

The file contains an LZW compressed TIFF file with plane size `64048 x 35029`, which the user is attempting to convert to pyramidal OME-TIFF. The original report showed the failure occur when using the below command:
`bfconvert -bigtiff -pyramid-resolutions 3 -pyramid-scale 3 -noflat /path/to/input out.ome.tiff`

The exact same failure can however be seen even without the pyramid or bigtiff options.
`bfconvert /path/to/input out.ome.tiff`

melissalinkert, 2025-10-17:
I have not so far been able to reproduce this problem with 6.11.0 or 8.3.0 and similarly-sized data, e.g. `'test&sizeX=64048&sizeY=35029&sizeC=3.fake'`.

I have requested access to the file in https://zenodo.org/records/7337371 (as it is not public), but if access isn't granted in time to look at this for 8.4.0 then plan to close this issue. We can always re-open later if more information becomes available.

melissalinkert, 2025-10-28:
I have not yet received access to https://zenodo.org/records/7337371, so closing in the absence of a way to test. It may be that #4058 is related or the same issue. We can re-open this if anyone else sees a similar problem and is able to provide public test data.