ome/bioformats #3918: Operetta Phenix: IllegalStateException when attempting to open plane

dgault, 2022-12-05:
Issue was reported on forum thread https://forum.image.sc/t/importing-opera-phenix-into-fiji-without-exported-images/73902/13

The original issue saw the below exception when attempting to open a large dataset on networked storage. The use provided the index xml which looked to be correct and valid. The issue was also consistently occurring and with multiple datasets so I do not believe it was network related. The user was using a Bio-Formats 6.10.1-SNAPSHOT version from 31 May 2022.
```
Caused by: java.lang.IllegalStateException: Could not open a plane for the given dimensions
	at io.scif.img.cell.loaders.AbstractArrayLoader.loadArray(AbstractArrayLoader.java:238)
	at io.scif.img.cell.SCIFIOCellImgFactory$SCIFIOCellLoader.load(SCIFIOCellImgFactory.java:200)
	at net.imglib2.cache.img.LoadedCellCacheLoader.get(LoadedCellCacheLoader.java:124)
	at net.imglib2.cache.img.LoadedCellCacheLoader.get(LoadedCellCacheLoader.java:84)
	at net.imglib2.cache.img.DiskCellCache.get(DiskCellCache.java:142)
	at net.imglib2.cache.img.DiskCellCache.get(DiskCellCache.java:81)
	at net.imglib2.cache.IoSync.get(IoSync.java:253)
	at net.imglib2.cache.ref.SoftRefLoaderRemoverCache.get(SoftRefLoaderRemoverCache.java:166)
	... 12 more
Caused by: java.io.IOException: Resetting to invalid mark
	at java.io.BufferedInputStream.reset(BufferedInputStream.java:448)
	at java.io.FilterInputStream.reset(FilterInputStream.java:226)
	at loci.common.URLHandle.seek(URLHandle.java:83)
	at loci.common.RandomAccessInputStream.seek(RandomAccessInputStream.java:203)
	at loci.formats.tiff.TiffParser.getTile(TiffParser.java:810)
	at loci.formats.tiff.TiffParser.getSamples(TiffParser.java:1116)
	at loci.formats.tiff.TiffParser.getSamples(TiffParser.java:871)
	at loci.formats.in.MinimalTiffReader.openBytes(MinimalTiffReader.java:312)
	at loci.formats.in.OperettaReader.openBytes(OperettaReader.java:245)
	at loci.formats.ImageReader.openBytes(ImageReader.java:465)
```

Furthermore a second error was also reported with a smaller dataset which suggests that the wrong read was being used for the fileset:
```
java.lang.IllegalArgumentException: Not a PVScan element

	at loci.formats.in.PrairieMetadata.checkElement(PrairieMetadata.java:428)

	at loci.formats.in.PrairieMetadata.parseXML(PrairieMetadata.java:239)

	at loci.formats.in.PrairieMetadata.<init>(PrairieMetadata.java:89)

	at loci.formats.in.PrairieReader.parsePrairieMetadata(PrairieReader.java:384)

	at loci.formats.in.PrairieReader.initFile(PrairieReader.java:349)

	at loci.formats.FormatReader.setId(FormatReader.java:1392)
```

This smaller sample file was also provided but I was unable to reproduce either of the reported errors with Bio-Formats 6.11.0 or 6.10.1. The sample file opened and displayed correctly without any errors or exceptions being thrown

melissalinkert, 2023-08-09:
https://github.com/ome/ome-common-java/pull/82 may help with the first issue.

The only way I can reproduce the second issue is by using `showinf -format Prairie Index.xml` or by turning off support for `PerkinElmer Operetta` in the ImageJ/Fiji `Plugins > Bio-Formats > Bio-Formats Plugins Configuration` window. Given that the original report used a snapshot version, it may be that there is either a configuration issue or a problem with conflicting update sites. If a known-good dataset (e.g. https://downloads.openmicroscopy.org/images/PerkinElmer-Operetta/omer/006P_M3/006P__2017-08-19T12_42_59-Measurement%203/Images/) shows a similar error, that would confirm that it's not a data problem.

dgault, 2023-10-06:
Closing this issue as the first reported problem has never been reproduced and the second issue has only been seen with an older SNAPSHOT version or by forcing an incorrect reader.

closed 2023-10-06 by dgault