ome/bioformats #3920: .TIFF from Napari readed in omero with inversion between T and Z

sodesset, 2022-12-06:
Hi,
I imported a binary image saved as tiff in napari by omero insight. 
In omero, the z axis is inversed with a time dimension.
The tif is read correctly by version 6.9.1 of bioformat, but I see the same inversion with version 6.11
Best regards

dgault, 2022-12-06:
Hi @sodesset, would it possible to upload the file for testing? If you need a suitable upload location then we recommend using https://zenodo.org/

sodesset, 2022-12-06:
Here is a file

-- 

*Sophie Desset*TEL  33 (0)4 73 40 53 04
Equipe CODED 
<https://www.gred-clermont.fr/directory/team/fr/equipe-03-dynamique-de-la-chromatine-et-du-noyau-3d/> 
– Plateforme CLIC 
<https://www.gred-clermont.fr/directory/platform/fr/clermont-ferrand-imagerie-confocale-clic/>

GReD– InsermU1103 – CNRS 6293 <https://www.gred-clermont.fr/>

UFR médecine
28 place Henri Dunant
TSA 50 400
63 001 Clermont-Ferrand CEDEX

sodesset, 2022-12-06:
Hi,

I just realized that in Omero/images details, we find the calibration of 
the image in the order z, y, x behind the argument {shape}: in the 
example of the screenshot corresponding to the file previously sent: 
{"shape": [63, 158, 144]}

Kind regards

-- 

*Sophie Desset*TEL  33 (0)4 73 40 53 04
Equipe CODED 
<https://www.gred-clermont.fr/directory/team/fr/equipe-03-dynamique-de-la-chromatine-et-du-noyau-3d/> 
– Plateforme CLIC 
<https://www.gred-clermont.fr/directory/platform/fr/clermont-ferrand-imagerie-confocale-clic/>

GReD– InsermU1103 – CNRS 6293 <https://www.gred-clermont.fr/>

UFR médecine
28 place Henri Dunant
TSA 50 400
63 001 Clermont-Ferrand CEDEX

dgault, 2022-12-06:
Hi @sodesset, we don't seem to have received the image you attached. Would you be able to upload to an online resource such as https://zenodo.org/

melissalinkert, 2023-08-09:
Closing; as @dgault noted above, there is simply not enough metadata in the file to distinguish between Z and T, so this isn't something we can fix in Bio-Formats. Opening in ImageJ, editing the Z/T dimensions, and resaving may be an option. You could also modify [this file conversion example code](https://github.com/ome/bio-formats-examples/blob/master/src/main/java/FileConvert.java) to explicitly set the Z and T dimensions for the output file (as is done e.g. [here](https://github.com/ome/bio-formats-examples/blob/master/src/main/java/FileExport.java#L141)).