ome/bioformats #3941: NDPI: IllegalArgumentException when parsing IFD values on Mac

dgault, 2023-02-07:
This issue was reported via imagesc thread https://forum.image.sc/t/some-ndpi-files-not-opening-in-qupath-v0-4-1/76590/24
A sample file has been provided which reproduces the error on Bio-Formats 6.11.1, this will shortly be copied to inbox

In the thread it was reported that the issue is present on Mac but that the file opens without issue on Windows. I have not yet been able to verify this on a Windows machine.

The exception occurs when parsing the Tiff IFD entries, 23 IFD’s in total, 3 channels, 7 resolutions and I assume a macro and label image. The stack trace below is seen when parsing one IFD entry which contains an offset which results in an IllegalArgumentException:

```
Exception in thread "main" java.lang.IllegalArgumentException
	at java.nio.Buffer.limit(Buffer.java:275)
	at loci.common.NIOFileHandle.read(NIOFileHandle.java:309)
	at loci.common.NIOFileHandle.read(NIOFileHandle.java:298)
	at loci.common.NIOFileHandle.read(NIOFileHandle.java:286)
	at loci.common.RandomAccessInputStream.read(RandomAccessInputStream.java:681)
	at loci.formats.tiff.TiffParser.getIFDValue(TiffParser.java:566)
	at loci.formats.tiff.TiffParser.fillInIFD(TiffParser.java:526)
	at loci.formats.in.MinimalTiffReader.initFile(MinimalTiffReader.java:483)
	at loci.formats.in.BaseTiffReader.initFile(BaseTiffReader.java:609)
	at loci.formats.in.NDPIReader.initFile(NDPIReader.java:311)
	at loci.formats.FormatReader.setId(FormatReader.java:1443)
	at loci.formats.ImageReader.setId(ImageReader.java:849)
```

melissalinkert, 2023-02-22:
I was able to get both files from the image.sc thread (I missed that a working download link was in a later post), and both have been copied to `inbox/gh-3941`. `showinf -nopix -noflat doesnt_work.ndpi` with Bio-Formats 6.11.1 on Windows throws an IllegalArgumentException as noted above. I was able to open `doesnt_work.ndpi` in QuPath 0.4.1 on Windows, but the OpenSlide image server is used, which implies that Bio-Formats failed to open it.

`works.ndpi` can be read with `showinf` and QuPath 0.4.1 using Bio-Formats. Both files can be opened in Hamamatsu's NDP.view2 software.

Something like this allows `doesnt_work.ndpi` to be initialized:

```
diff --git a/components/formats-gpl/src/loci/formats/in/NDPIReader.java b/components/formats-gpl/src/loci/formats/in/NDPIReader.java
index c2db302920..1ee6ca7add 100644
--- a/components/formats-gpl/src/loci/formats/in/NDPIReader.java
+++ b/components/formats-gpl/src/loci/formats/in/NDPIReader.java
@@ -305,10 +305,10 @@ public class NDPIReader extends BaseTiffReader {
   /* @see loci.formats.FormatReader#initFile(String) */
   @Override
   protected void initFile(String id) throws FormatException, IOException {
+    super.initFile(id);
     RandomAccessInputStream s = new RandomAccessInputStream(id);
     use64Bit = s.length() >= Math.pow(2, 32);
     s.close();
-    super.initFile(id);
   }
 
   // -- Internal BaseTiffReader API methods --
```

but that will cause problems for working files larger than 4GB and doesn't fix all image reading for `doesnt_work.ndpi`, so a more targeted solution is required.

dgault, 2023-09-14:
An additional report of a similar issue has been reported on https://forum.image.sc/t/problems-uploading-ndpi-files-using-insight/86248/7

dgault, 2023-09-18:
Sample files from the latest report are available at https://zenodo.org/record/8344641 and `scratch/inbox/mwicks-ndpi-import-fail`

PR #4099 (2023-10-04): TIFF: check IFD entry offset before seeking
Fixes #3941 (I think). `showinf -noflat -resolution 6` on each of 3 affected files referenced in #3941 works with this PR, but does not work without this PR. The forthcoming configuration PR will include NDP.view2 screenshots for reference; it's worth noting that all 3 affected files have 3 Z sections.

Assuming tests pass, should be safe for a patch release.
commit fc6e42ee (2023-10-04): TIFF: check IFD entry offset before seeking

...when using artifical 64-bit offsets. This is identical to the check that
occurs when using the file's native offset width.

Fixes #3941
closed 2023-10-04 by dgault