ome/bioformats #3944: VSI

jboulanger, 2023-02-13:
Hello,
We have an issue with decoding VSI files from Evident's VS200 slide scanner after an update of cellsens software.

Here is the trace of the error when loading an image in FijiJ
(Fiji Is Just) ImageJ 2.9.0/1.53t; Java 1.8.0_322 [64-bit]; Linux 5.4.0-137-generic; 117MB of 5610MB (2%)
 
java.lang.IndexOutOfBoundsException: Index: 8, Size: 8
	at java.util.ArrayList.rangeCheck(ArrayList.java:659)
	at java.util.ArrayList.get(ArrayList.java:435)
	at loci.formats.in.CellSensReader.parseETSFile(CellSensReader.java:1207)
	at loci.formats.in.CellSensReader.initFile(CellSensReader.java:742)
	at loci.formats.FormatReader.setId(FormatReader.java:1443)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:499)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:142)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:237)
	at ij.IJ.runPlugIn(IJ.java:203)
	at ij.IJ.runPlugIn(IJ.java:192)
	at HandleExtraFileTypes.openImage(HandleExtraFileTypes.java:524)
	at HandleExtraFileTypes.run(HandleExtraFileTypes.java:97)
	at ij.IJ.runUserPlugIn(IJ.java:237)
	at ij.IJ.runPlugIn(IJ.java:203)
	at ij.IJ.runPlugIn(IJ.java:192)
	at ij.io.Opener.openWithHandleExtraFileTypes(Opener.java:555)
	at ij.io.Opener.openUsingHandleExtraFileTypes(Opener.java:407)
	at ij.io.Opener.openTiff(Opener.java:913)
	at ij.io.Opener.openImage(Opener.java:335)
	at ij.io.Opener.openImage(Opener.java:241)
	at ij.io.Opener.open(Opener.java:104)
	at ij.io.Opener.open(Opener.java:73)
	at ij.plugin.Commands.run(Commands.java:27)
	at ij.IJ.runPlugIn(IJ.java:209)
	at ij.Executer.runCommand(Executer.java:152)
	at ij.Executer.run(Executer.java:70)
	at java.lang.Thread.run(Thread.java:750)

sbesson, 2023-02-13:
Hi @jboulanger, the stack trace and the acquisition software upgrade make me suspicious this is another occurrence of https://github.com/ome/bioformats/issues/3859. 

In this case, I would expect https://github.com/ome/bioformats/pull/3925 to include a fix allowing to read files produced by the latest Evident slide scanner upgrade. Would you have any sample data that you could share allowing us to confirm this?

sbesson, 2023-03-06:
Closing as Bio-Formats 6.12.0 should address the reading of VSI filesets produced by Evident's VS200 scanner.

closed 2023-03-06 by sbesson