ome/bioformats #3949: Less verbose output when loading multifile TIFFs

anntzer, 2023-02-23:
When converting a multifile TIFF (e.g. from a Metamorph .nd MDA acquisition) with bfconvert (or when opening it with fiji), the output can get many, many lines of "Reading IFDs"/"Populating metadata"/"Populating OME metadata", which is in fact obscuring the (more) interesting output (e.g. the progress from bfconvert "Series S: converted T/U planes (V%)").  Naturally, one can grep these repeated lines away, but perhaps consider not emitting them at all, or making them togglable via a command-line switch or environment variable?

dgault, 2023-02-24:
Hi @anntzer, thank you for the feedback. The `bfconvert` tool in particular currently only has an option (-debug) to toggle the debug logging output, so you can't currently disable the output you are seeing. The `showinf` tool has a greater number of options for different levels of debugging (debug, info, trace and off). We should be able to add similar options to the bfconvert tool to give a greater level of control to the output.

anntzer, 2023-04-17:
From a quick look, both the "Reading IFDs"/... logs and the progress report logs are made at INFO level, so I'm not sure a global switch would be enough (it's still better than nothing, though).

dgault, 2023-04-18:
Yeah I could see the progress reports being moved to a different level in this case.

closed 2026-06-12 by melissalinkert