ome/bioformats #3957: BDV: Remove absolute file paths

dgault, 2023-03-06:
This issue was raised on https://github.com/BIOP/ijp-imagetoatlas/issues/154#issuecomment-1420533141

In this scenario a memo file has been created, which contains an absolute file path (the h5Id variable). When the dataset along with the memo file are moved to a different location then the file fails to open as the settings stored in the memo file are invalid.

NicoKiaru, 2023-09-21:
Just a heads-up that this problem also occurs with TIFFs.

carlocastoldi, 2026-01-21:
Interestingly, using relative paths, when possible, it should be an easy fix. It's a matter to working on the `directory` field of the `Memoizer`, and close to this line:
https://github.com/ome/bioformats/blob/eb8fd308b0a1004596899247bc0505b9ee6a97f4/components/formats-bsd/src/loci/formats/Memoizer.java#L832