ome/bioformats #3961: Import failure of Ionpath MIBITIFF with ImageJ and Bio-Formats 6.10 / 6.12 (current)

christianrickert, 2023-03-09:
There might be a issue with Bio-Formats when importing of Ionpath MIBITIFF-compatible files in ImageJ 1.54c (Java 1.8.0_322, 64-bit):

Example: [[FOV1-1] MoQC-100.zip](https://github.com/ome/bioformats/files/10935583/FOV1-1.MoQC-100.zip)

[image]

Bio-Formats 6.10.1  (plugins/bio-formats_plugins-6.10.1-20220731.012559-15.jar), from [Bio-Formats](https://sites.imagej.net/Bio-Formats/) in ImageJ Updater
```
loci.formats.FormatException: Unexpected format in SIMS description JSON.
	at loci.formats.in.IonpathMIBITiffReader.initStandardMetadata(IonpathMIBITiffReader.java:142)
	at loci.formats.in.BaseTiffReader.initMetadata(BaseTiffReader.java:98)
	at loci.formats.in.BaseTiffReader.initFile(BaseTiffReader.java:610)
	at loci.formats.FormatReader.setId(FormatReader.java:1443)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:499)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:142)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:244)
	at ij.IJ.runPlugIn(IJ.java:210)
	at ij.IJ.runPlugIn(IJ.java:199)
	at loci.plugins.shortcut.ShortcutPanel$1.run(ShortcutPanel.java:189)
```

Bio-Formats 6.11.1 (plugins/bio-formats_plugins-6.11.1.jar), currently ships with Fiji
```
(no issue)
```

Bio-Formats 6.12.0 (plugins/bio-formats_plugins-6.12.0.jar), updated by ImageJ Updater's default sites
```
java.lang.IllegalStateException: Too early in import process: current step is FILE, but must be after STACK
	at loci.plugins.in.ImportProcess.assertStep(ImportProcess.java:778)
	at loci.plugins.in.ImportProcess.getReader(ImportProcess.java:306)
	at loci.plugins.in.Importer.run(Importer.java:99)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:244)
	at ij.IJ.runPlugIn(IJ.java:210)
	at ij.IJ.runPlugIn(IJ.java:199)
	at loci.plugins.shortcut.ShortcutPanel$1.run(ShortcutPanel.java:189)
```

The [latest commit](https://github.com/ome/bioformats/commit/b1e0910f3c00a923d8e12f73881e618ff4691bb8) in [IonpathMIBITiffReader.java](https://github.com/ome/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/IonpathMIBITiffReader.java) was on September 17, 2018.

Independent from the specific file type, I can see the `step is FILE, but must be after STACK` error being described earlier [image.sc](https://forum.image.sc):
[Cannot import .czi stacks to FIJI](https://forum.image.sc/t/cannot-import-czi-stacks-to-fiji/73220)
[Some .ndpi files not opening in Qupath v0.4.1](https://forum.image.sc/t/some-ndpi-files-not-opening-in-qupath-v0-4-1/76590)
[Exception Issue when opening .nd2 files with Bio-Formats](https://forum.image.sc/t/exception-issue-when-opening-nd2-files-with-bio-formats/40462)
[Error importing .czi videos](https://forum.image.sc/t/error-importing-czi-videos/73314)

dgault, 2023-03-17:
Hi @christianrickert, thank you for reporting the issue and providing aa sample file. This looks to be an odd issue as the IonpathMIBITiffReader hasn't changed in quite some time. I tested the sample file with Bio-Formats 6.12.0 and was unable to reproduce the exception. When looking into the metadata in the file, it is always using `"image.type":"SIMS"` which is correct and shouldn't produce any issue.

The stack trace from the ImageJ plugin is perhaps a bit misleading as you will see the same stack for many different types of exceptions caught in the plugin.

Since the version that worked for you was a version that shipped directly with FIJI, I suspect the issue may be with the dependencies in your FIJI environment when modifying it from the update site. Could you list the names of jars in your `FIJI/jars` folder and also `FIJI/jars/bio-formats`?

dgault, 2023-03-20:
Thanks @christianrickert, it does indeed seem to be related to the json jar. I was able to reproduce the issue you are seeing with the json-20220924 jar. The issue seems to occur when [reading the `channel.mass` value](https://github.com/ome/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/IonpathMIBITiffReader.java#L133) from the JSON description, with previous versions of the jar this returned a String value but with the latest 20220924 version it is now returning a double value. 

The reader will need to be updated to handle both of the possible return values. Until that is updated you would be best reverting back to the `json-20090211.jar` if possible.

dgault, 2023-03-21:
Yeah we will certainly follow up with the ImageJ team on this. Firstly I have opened a PR to look at bumping the json dependency on our side and that will let me see exactly what the impact is and how many readers will be affected - https://github.com/ome/bioformats/pull/3966

christianrickert, 2023-03-21:
Much appreciated! 🤝

dgault, 2023-06-23:
Closing as this has been resolved with the dependency updates in 6.13.0

closed 2023-06-23 by dgault