ome/bioformats #3981: STK: use CreateTime as alternate AcquisitionDate

anntzer, 2023-04-17:
STK tiffs do not record the acquisition date (AFAICT), but have a CreateTime tag which could be used instead.  On the files I have (generated by Metamorph), the TIFF standard DateTime tag is also filled with the same value.  Currently, it looks like bioformats just ignores that information; I would like to kindly request that CreateTime be used instead in that case.  (This seems conceptually similar to #3430.)

Example:

movie.nd:
```
"NDInfoFile", Version 1.0
"Description",
"StartTime1", 20230414 16:27:04
"DoTimelapse", TRUE
"NTimePoints", 2
"DoStage", TRUE
"NStagePositions", 1
"Stage1", "stage1"
"DoWave", TRUE
"NWavelengths", 1
"WaveName1", "491"
"WaveDoZ1", FALSE
"DoZSeries", FALSE
"WaveInFileName", TRUE
"NEvents", 0
"EndFile"
```
Python script to recreate tiff files (the metadata is from real files, I simply replaced the actual data segment by zeros to have a reasonably-sized script):
```python
import base64
header = b'II*\x00\xac\t\x08\x00'
footers = list(map(base64.b64decode, [
    b'RXhwb3N1cmU6IDEwMCBtcw0KQmlubmluZzogMSB4IDENClJlZ2lvbjogNTEyIHggNTEyLCBvZmZzZXQgYXQgKDAsIDApDQpTdWJ0cmFjdDogT2ZmDQpTaGFkaW5nOiBPZmYNCkRpZ2l0aXplcjogMTAgTUh6IChFTSBHYWluKQ0KR2FpbjogR2FpbiAzICg0eCkNCk11bHRpcGxpY2F0aW9uIEdhaW46IDMwMA0KQ2FtZXJhIFNodXR0ZXI6IEFsd2F5cyBPcGVuDQpDbGVhciBDb3VudDogMg0KQ2xlYXIgTW9kZTogQ0xFQVIgUFJFIFNFUVVFTkNFDQpGcmFtZXMgdG8gQXZlcmFnZTogMQ0KVHJpZ2dlciBNb2RlOiBOb3JtYWwgKFRJTUVEKQ0KUHJvY2Vzc2luZyBNZXRob2Q6IE5vcm1hbA0KQmxhY2sgTG9jayBFbmFibGVkOiBObw0KVG9wIExvY2sgRW5hYmxlZDogTm8NCkRlc3BlY2tsZSBFbmFibGVkOiBObw0KVGVtcGVyYXR1cmU6IC04MC4xMw0KSWxsdW1pbmF0aW9uOiA0OTENCg0KAAgAAAAIIAAACEAAAAhgAAAIgAAACKAAAAjAAAAI4AAACAABAAggAQAIQAEACGABAAiAAQAIoAEACMABAAjgAQAIAAIACCACAAhAAgAIYAIACIACAAigAgAIwAIACOACAAgAAwAIIAMACEADAAhgAwAIgAMACKADAAjAAwAI4AMACAAEAAggBAAIQAQACGAEAAiABAAIoAQACMAEAAjgBAAIAAUACCAFAAhABQAIYAUACIAFAAigBQAIwAUACOAFAAgABgAIIAYACEAGAAhgBgAIgAYACKAGAAjABgAI4AYACAAHAAggBwAIQAcACGAHAAiABwAIoAcACMAHAAjgBwAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAASAAAAAEAAABIAAAAAQAAAE1ldGFNb3JwaCA3LjguMy4wADIwMjM6MDQ6MTQgMTY6Mjc6MTEAAAEAAAABAAAAAQAAAAEAAAAGAAAAcGl4ZWwABAAAADQ5MQCQiSUAGcuHA5CJJQAdzYcD/////wEAAAD/////AQAAAL45AAABAAAAvc///woAAAAAAAAAAQAAAAAAAAABAAAADAAAAG15bzUtaGFsZi0xAAUAAABkAAAABQAAAGQAAADANRIMQEIPAAEAAAABAAAAAQAAAAEAAAAAAAAADl9JbGx1bVNldHRpbmdfAgAAAAIDNDkxAAAAAAdfTWFnTkFfAgAAAAEAAAAAAQAAAAAAAAAHX01hZ1JJXwIAAAABAAAAAAEAAAAAAAAADF9NYWdTZXR0aW5nXwIAAAACAAAAAAAQQ2FtZXJhIEJpdCBEZXB0aAAAAAABEAAAAAEAAAAAAAAAG0REQTAyIENIIFNodXR0ZXIgMSBCbGFua2luZwAAAAACBkNsb3NlZAAAAAAbRERBMDIgQ0ggU2h1dHRlciA0IFNlY3VyaXTpAAAAAAIGQ2xvc2VkAAAAABtEREEwMiBJbnRlbnNpdHkgU2h1dHRlciA0MDUAAAAAAgZDbG9zZWQAAAAAG0REQTAyIEludGVuc2l0eSBTaHV0dGVyIDQ5MQAAAAACBkNsb3NlZAAAAAAbRERBMDIgSW50ZW5zaXR5IFNodXR0ZXIgNTYxAAAAAAIGQ2xvc2VkAAAAABtEREEwMiBJbnRlbnNpdHkgU2h1dHRlciA2NDIAAAAAAgZDbG9zZWQAAAAAFkxhbWJkYSBTaHV0dGVyIFRSQU5TIEEAAAAAAgZDbG9zZWQAAAAAE0xhc2VyX0ludGVuc2l0eV80MDUAAAAAAQAAAAABAAAAAAAAABNMYXNlcl9JbnRlbnNpdHlfNDkxAAAAAAEyAAAAAQAAAAAAAAATTGFzZXJfSW50ZW5zaXR5XzU2MQAAAAABBQAAAAEAAAAAAAAAE0xhc2VyX0ludGVuc2l0eV82NDIAAAAAAQAAAAABAAAAAAAAAAxUaSBDb25kZW5zZXIAAAAAAgUtLS0tLQAAAAARVGkgRmlsdGVyIEJsb2NrIDEAAAAAAgUtLS0tLQAAAAANVGkgTGFtcCBQb3dlcgAAAAACBE9wZW4AAAAADFRpIE9iamVjdGl2ZQAAAAACGVBsYW4gQXBvIFRJUkYgMTAweCAvIDEuNDUAAAAAD1RpIE9wdGljYWwgUGF0aAAAAAACCUxlZnQgUG9ydAAAAAAHVGkgVElSRgAAAAACBFRJUkYAAAAADVRpIFRJUkYgQW5nbGUAAAAAAaoFAAABAAAAEgAAAFNldCBCeSBXYXZlbGVuZ3RoAAAAAAABAAAAAQAAAEgUAAACAAAAwP8AAAMAAAABAAAABAAAAPkDCAAFAAAAAQQIAAYAAAAJBAgABwAAABMECAAIAAAAAAAAAAkAAAD/AAAACwAAAIAAAAAMAAAAQAAAAA0AAAAAAAAADgAAAP//AAAPAAAA0gAAABAAAAAbBAgAEQAAACMECAATAAAABAAAABQAAAAAAAAAFQAAACsECAAWAAAAKwQIABcAAAArBAgAGAAAADMECAAaAAAAAAAAABsAAAAAAAAAHAAAADsECAAdAAAASwQIACUAAABbBAgAJgAAAGsECAAnAAAAcwQIACgAAAB7BAgAKQAAAAAAAAAqAAAAgwQIAC4AAACLBAgALwAAAAgAAAAxAAAAkwQIADEAAACvBAgAMQAAAMgECAAxAAAA4QQIADEAAAD4BAgAMQAAABoFCAAxAAAARgUIADEAAAByBQgAMQAAAJ4FCAAxAAAAygUIADEAAAD2BQgAMQAAACIGCAAxAAAASQYIADEAAABuBggAMQAAAJMGCAAxAAAAuAYIADEAAADdBggAMQAAAPkGCAAxAAAAGgcIADEAAAA2BwgAMQAAAGYHCAAxAAAAiQcIADEAAACfBwgAQwAAAL4HCAAUAP4ABAABAAAAAAAAAAABBAABAAAAAAIAAAEBBAABAAAAAAIAAAIBAwABAAAAEAAAAAMBAwABAAAAAQAAAAYBAwABAAAAAQAAAAoBAwABAAAAAQAAAA4BAgC6AQAACAAIABEBBABAAAAAwgEIABUBAwABAAAAAQAAABYBBAABAAAACAAAABcBBABAAAAAwgIIABoBBQABAAAAwgMIABsBBQABAAAAygMIABwBAwABAAAAAQAAACgBAwABAAAAAgAAADEBAgASAAAA0gMIADIBAgAVAAAA5AMIAD0BAwABAAAAAQAAAFyDBAA7AAAA1AcIAAAAAAA=',
    b'RXhwb3N1cmU6IDEwMCBtcw0KQmlubmluZzogMSB4IDENClJlZ2lvbjogNTEyIHggNTEyLCBvZmZzZXQgYXQgKDAsIDApDQpTdWJ0cmFjdDogT2ZmDQpTaGFkaW5nOiBPZmYNCkRpZ2l0aXplcjogMTAgTUh6IChFTSBHYWluKQ0KR2FpbjogR2FpbiAzICg0eCkNCk11bHRpcGxpY2F0aW9uIEdhaW46IDMwMA0KQ2FtZXJhIFNodXR0ZXI6IEFsd2F5cyBPcGVuDQpDbGVhciBDb3VudDogMg0KQ2xlYXIgTW9kZTogQ0xFQVIgUFJFIFNFUVVFTkNFDQpGcmFtZXMgdG8gQXZlcmFnZTogMQ0KVHJpZ2dlciBNb2RlOiBOb3JtYWwgKFRJTUVEKQ0KUHJvY2Vzc2luZyBNZXRob2Q6IE5vcm1hbA0KQmxhY2sgTG9jayBFbmFibGVkOiBObw0KVG9wIExvY2sgRW5hYmxlZDogTm8NCkRlc3BlY2tsZSBFbmFibGVkOiBObw0KVGVtcGVyYXR1cmU6IC03OS45NA0KSWxsdW1pbmF0aW9uOiA0OTENCg0KAAgAAAAIIAAACEAAAAhgAAAIgAAACKAAAAjAAAAI4AAACAABAAggAQAIQAEACGABAAiAAQAIoAEACMABAAjgAQAIAAIACCACAAhAAgAIYAIACIACAAigAgAIwAIACOACAAgAAwAIIAMACEADAAhgAwAIgAMACKADAAjAAwAI4AMACAAEAAggBAAIQAQACGAEAAiABAAIoAQACMAEAAjgBAAIAAUACCAFAAhABQAIYAUACIAFAAigBQAIwAUACOAFAAgABgAIIAYACEAGAAhgBgAIgAYACKAGAAjABgAI4AYACAAHAAggBwAIQAcACGAHAAiABwAIoAcACMAHAAjgBwAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAAACAAAAAgAAAAIAAASAAAAAEAAABIAAAAAQAAAE1ldGFNb3JwaCA3LjguMy4wADIwMjM6MDQ6MTQgMTY6Mjg6NTIAAAEAAAABAAAAAQAAAAEAAAAGAAAAcGl4ZWwABAAAADQ5MQCQiSUA5laJA5CJJQB8WIkD/////wEAAAD/////AQAAAL45AAABAAAAvc///woAAAAAAAAAAQAAAAAAAAABAAAADAAAAG15bzUtaGFsZi0xAAUAAABkAAAABQAAAGQAAAAXTwAAZAAAAAEAAAABAAAAAQAAAAEAAAAAAAAADl9JbGx1bVNldHRpbmdfAgAAAAIDNDkxAAAAAAdfTWFnTkFfAgAAAAEAAAAAAQAAAAAAAAAHX01hZ1JJXwIAAAABAAAAAAEAAAAAAAAADF9NYWdTZXR0aW5nXwIAAAACAAAAAAAQQ2FtZXJhIEJpdCBEZXB0aAAAAAABEAAAAAEAAAAAAAAAG0REQTAyIENIIFNodXR0ZXIgMSBCbGFua2luZwAAAAACBkNsb3NlZAAAAAAbRERBMDIgQ0ggU2h1dHRlciA0IFNlY3VyaXTpAAAAAAIGQ2xvc2VkAAAAABtEREEwMiBJbnRlbnNpdHkgU2h1dHRlciA0MDUAAAAAAgZDbG9zZWQAAAAAG0REQTAyIEludGVuc2l0eSBTaHV0dGVyIDQ5MQAAAAACBkNsb3NlZAAAAAAbRERBMDIgSW50ZW5zaXR5IFNodXR0ZXIgNTYxAAAAAAIGQ2xvc2VkAAAAABtEREEwMiBJbnRlbnNpdHkgU2h1dHRlciA2NDIAAAAAAgZDbG9zZWQAAAAAFkxhbWJkYSBTaHV0dGVyIFRSQU5TIEEAAAAAAgZDbG9zZWQAAAAAE0xhc2VyX0ludGVuc2l0eV80MDUAAAAAAQAAAAABAAAAAAAAABNMYXNlcl9JbnRlbnNpdHlfNDkxAAAAAAEyAAAAAQAAAAAAAAATTGFzZXJfSW50ZW5zaXR5XzU2MQAAAAABBQAAAAEAAAAAAAAAE0xhc2VyX0ludGVuc2l0eV82NDIAAAAAAQAAAAABAAAAAAAAAAxUaSBDb25kZW5zZXIAAAAAAgUtLS0tLQAAAAARVGkgRmlsdGVyIEJsb2NrIDEAAAAAAgUtLS0tLQAAAAANVGkgTGFtcCBQb3dlcgAAAAACBE9wZW4AAAAADFRpIE9iamVjdGl2ZQAAAAACGVBsYW4gQXBvIFRJUkYgMTAweCAvIDEuNDUAAAAAD1RpIE9wdGljYWwgUGF0aAAAAAACCUxlZnQgUG9ydAAAAAAHVGkgVElSRgAAAAACBFRJUkYAAAAADVRpIFRJUkYgQW5nbGUAAAAAAaoFAAABAAAAEgAAAFNldCBCeSBXYXZlbGVuZ3RoAAAAAAABAAAAAQAAAOoTAAACAAAArP8AAAMAAAABAAAABAAAAPkDCAAFAAAAAQQIAAYAAAAJBAgABwAAABMECAAIAAAAAAAAAAkAAAD/AAAACwAAAIAAAAAMAAAAQAAAAA0AAAAAAAAADgAAAP//AAAPAAAA0gAAABAAAAAbBAgAEQAAACMECAATAAAABAAAABQAAAAAAAAAFQAAACsECAAWAAAAKwQIABcAAAArBAgAGAAAADMECAAaAAAAAAAAABsAAAAAAAAAHAAAADsECAAdAAAASwQIACUAAABbBAgAJgAAAGsECAAnAAAAcwQIACgAAAB7BAgAKQAAAAAAAAAqAAAAgwQIAC4AAACLBAgALwAAAAgAAAAxAAAAkwQIADEAAACvBAgAMQAAAMgECAAxAAAA4QQIADEAAAD4BAgAMQAAABoFCAAxAAAARgUIADEAAAByBQgAMQAAAJ4FCAAxAAAAygUIADEAAAD2BQgAMQAAACIGCAAxAAAASQYIADEAAABuBggAMQAAAJMGCAAxAAAAuAYIADEAAADdBggAMQAAAPkGCAAxAAAAGgcIADEAAAA2BwgAMQAAAGYHCAAxAAAAiQcIADEAAACfBwgAQwAAAL4HCAAUAP4ABAABAAAAAAAAAAABBAABAAAAAAIAAAEBBAABAAAAAAIAAAIBAwABAAAAEAAAAAMBAwABAAAAAQAAAAYBAwABAAAAAQAAAAoBAwABAAAAAQAAAA4BAgC6AQAACAAIABEBBABAAAAAwgEIABUBAwABAAAAAQAAABYBBAABAAAACAAAABcBBABAAAAAwgIIABoBBQABAAAAwgMIABsBBQABAAAAygMIABwBAwABAAAAAQAAACgBAwABAAAAAgAAADEBAgASAAAA0gMIADIBAgAVAAAA5AMIAD0BAwABAAAAAQAAAFyDBAA7AAAA1AcIAAAAAAA='
]))
for t in [1, 2]:
    with open(f"movie_w1491_s1_t{t}.TIF", "wb") as file:
        file.write(header)
        file.write(512 * 512 * 2 * b"\0")
        file.write(footers[t - 1])
```
After generating the tiff files, check e.g. with `python -mtifffile movie_w1491_s1_t1.TIF` that CreateTime/DateTime are correctly set; however, `bfconvert movie.nd movie.ome.tiff` creates an OME tiff with DeltaT set to zero.

dgault, 2023-04-18:
For Metamorph files such as these, it will be the `StartTime1` tag in the ND file which is getting used to set the acquisition date in Bio-Formats.

When I run your example the acquisition date still seems to populated in the resulting OME-TIFF as below:
```
   <Image ID="Image:0" Name="Stage1">
       <AcquisitionDate>2023-04-14T16:27:04</AcquisitionDate>
```

Are you not seeing the acquisition date? Or have I misunderstood which CreateTime tag you are referring to?

dgault, 2023-04-19:
Thanks, that makes sense now.