ome/bioformats #4015: CellSens VSI without pyramid

jboulanger, 2023-06-09:
Hello,
I have a VSI file exported from the Evident software which is made of a single vsi file (with no ets pyramid) as it is only a few kB or MB.
In that case, bioformats does not seem to decode properly the image serie but only the macro image.
Here is a picture https://cloud.mrc-lmb.cam.ac.uk/s/PT5FeBsxkMkabxy
I hope this helps.
All the best,

dgault, 2023-06-12:
Thanks @jboulanger, with the latest Bio-Formats it seems to have picked up 2 series, both labeled as macro image, the first 152 x 167 (3 channels) and the second 62 x 68 (5 channels). The second does not display correctly and is scrambled.

dgault, 2024-06-04:
A similar issue has been reported in forum thread https://forum.image.sc/t/open-vsi-ome-tiff-files-with-tif-pixel-data/96327/9

A sample dataset acquired from Olympus VS200 has been provided and reproduces the issue with Bio-Formats 7.3.0. The samples files are at https://drive.google.com/drive/folders/10PX3ivPcWImZJF2DYKsiztp0cHM_ZZIu and I will copy them over to inbox for testing

In the dataset, it includes a `.tiff` file which appears to have multiple resolutions and has no associated .ets file. 

When opening the .vsi file (in this case fluorescent.vsi from the sample data), we have 3 series as below, the last of which seems to be the lowest level resolution:
```
Series count = 3
Series #0 :
	Image count = 1
	RGB = true (3) 
	Interleaved = false
	Indexed = false (true color)
	Width = 991
	Height = 375
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 991 x 375
	Thumbnail size = 128 x 48
	Endianness = intel (little)
	Dimension order = XYCZT (uncertain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #1 :
	Image count = 1
	RGB = true (3) 
	Interleaved = false
	Indexed = false (true color)
	Width = 495
	Height = 284
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 495 x 64
	Thumbnail size = 128 x 73
	Endianness = intel (little)
	Dimension order = XYCZT (uncertain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = true
	Thumbnail series = true
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #2 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (true color)
	Width = 454
	Height = 374
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Tile size = 454 x 64
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (uncertain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = true
	-----
	Plane #0 <=> Z 0, C 0, T 0
```

When opening the tiff file on its own (frame_t_0-001.tiff from the sample data) it is read as an OME-TIFF file, but even with the noflat option each of the resolutions appears as a seperate series:
```
Series count = 7
Reading IFDs
Populating metadata
Series #0 :
	Image count = 3
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 29074
	Height = 23985
	SizeZ = 1
	SizeT = 1
	SizeC = 3
	Tile size = 512 x 512
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0
	Plane #2 <=> Z 0, C 2, T 0

Series #1 :
	Image count = 3
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 14537
	Height = 11992
	SizeZ = 1
	SizeT = 1
	SizeC = 3
	Tile size = 512 x 512
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0
	Plane #2 <=> Z 0, C 2, T 0

Series #2 :
	Image count = 3
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 7268
	Height = 5996
	SizeZ = 1
	SizeT = 1
	SizeC = 3
	Tile size = 512 x 512
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0
	Plane #2 <=> Z 0, C 2, T 0

Series #3 :
	Image count = 3
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 3634
	Height = 2998
	SizeZ = 1
	SizeT = 1
	SizeC = 3
	Tile size = 512 x 512
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0
	Plane #2 <=> Z 0, C 2, T 0

Series #4 :
	Image count = 3
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 1817
	Height = 1499
	SizeZ = 1
	SizeT = 1
	SizeC = 3
	Tile size = 512 x 512
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0
	Plane #2 <=> Z 0, C 2, T 0

Series #5 :
	Image count = 3
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 908
	Height = 749
	SizeZ = 1
	SizeT = 1
	SizeC = 3
	Tile size = 512 x 512
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0
	Plane #2 <=> Z 0, C 2, T 0

Series #6 :
	Image count = 3
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 454
	Height = 374
	SizeZ = 1
	SizeT = 1
	SizeC = 3
	Tile size = 454 x 374
	Thumbnail size = 128 x 105
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #1 <=> Z 0, C 1, T 0
	Plane #2 <=> Z 0, C 2, T 0

```

closed 2026-09-17 by sbesson