ome/bioformats #4017: Convert scn file to jpg error

SinnoSong, 2023-06-14:
I want convert `.scn` file to `.jpg`,
```
FormatTools.convert(inputPath, outputPath);
```
, here is the exception:
```
Exception in thread "main" loci.formats.FormatException: Unsupported image type 'uint16'.
	at loci.formats.FormatWriter.checkParams(FormatWriter.java:482)
	at loci.formats.out.ImageIOWriter.saveBytes(ImageIOWriter.java:87)
	at loci.formats.FormatWriter.saveBytes(FormatWriter.java:132)
	at loci.formats.ImageWriter.saveBytes(ImageWriter.java:252)
	at loci.formats.FormatTools.convert(FormatTools.java:1484)
	at loci.formats.FormatTools.convert(FormatTools.java:1436)
	at com.yxys.Main.main(Main.java:15)
```

dgault, 2023-06-14:
Hi @SinnoSong, the base JPEG Writer in Bio-Formats only supports UINT8 data, if a JPEG2000 file would suit you can instead use the jp2 file extension, this should allow you to write your data. Alternatively, converting to a TIFF or OME-TIFF will also let you use JPEG compression if that is an option.

SinnoSong, 2023-06-15:
@dgault Thanks!

closed 2023-06-15 by SinnoSong