ome/bioformats #4021: Metamorph reader reads .nd files using wrong encoding

anntzer, 2023-06-15:
Currently, the metamorph nd reader reads nd files as utf-8 (essentially, in MetamorphReader.java, initFile() calls `DataTools.readFile(ndFileName)`; readFile is implemented in ome-common-java/.../DataTools.java and uses Constants.ENCODING = "UTF-8" from ome-common-java/.../Constants.java.

But generating MDA acquisitions on Metamorph (v7.10.5.476 on Windows) with non-ascii stage position names shows that that encoding is incorrect: I believe that the correct encoding is actually Windows-1252, though I cannot find documentation that explicitly states that.  (In particular, € is encoded at 0x80, which distinguishes the encoding from latin-1, and trying to input some non-Windows-1252 characters in position names results in these being replaced by question marks.)  In practice, such characters currently end up being replaced by a Unicode replacement character (U+FFFD �).

I would therefore suggest fixing the encoding, e.g. by providing a `DataTools.readFile(String id, Charset charset)` overload and using `readFile(ndFilename, "windows-1252")` in MetamorphReader.java.

dgault, 2023-06-16:
Hi @anntzer, thank you for raising the issue and investigating the problem. Making DataTools.readFile more flexible to also allow for the charset to be input seems like a good solution for this, it would allow for the MetaMorph and in future other readers to have more control of the reading. Do you think you would be able to open a Pull Request with that change? If you need any help or pointers I am happy to assist.

Also do you have a sample file you could share with the Windows-1252 encoding that we could use for testing? If you need a suitable upload location for a sample file, we recommend using https://zenodo.org/

anntzer, 2023-06-16:
Here is a test nd file (zipped because github allows zip uploads but not nd uploads...) [test.nd.zip](https://github.com/ome/bioformats/files/11772469/test.nd.zip)
The Stage1 position name should read "µ€œ?" (actually "µ€œẞ" but Metamorph immediately replaces the non-Windows1252 "ẞ" by at "?" at the GUI-level.)

I can try posting a tentative patch, but having no experience with java or in particular with java build tools I don't think it would be particularly efficient for me to actually post a PR (especially given that this looks like it'll need two PRs, one to ome-common-java and one to bioformats).

dgault, 2023-06-19:
Thanks @anntzer, I will try to get those opened as PR's in the next few days

sbesson, 2025-08-01:
@anntzer we are looking into fixing this issue as part of the upcoming Bio-Formats 8.3.0 release and planning to add your representative sample to our curated QA repository for non-regression testing. Consistently with our [contribution policy](https://ome-contributing.readthedocs.io/en/latest/third-party-policy.html#receipt-of-sample-data), we would like to make it part of our publicly available samples, exposed [on the OME downloads site](https://downloads.openmicroscopy.org/images/Metamorph/) for this file format. Would you be able to re-submit the sample data to the [Zenodo Bio-Formats community](https://zenodo.org/communities/bio-formats/records?q=&l=list&p=1&s=10&sort=newest) so that it can be labelled with the appropriate attribution and permissive licensing information?

anntzer, 2025-08-01:
My understanding is that you only need the .nd file from myself, right?  If so I hereby place that file (test.nd mentioned at https://github.com/ome/bioformats/issues/4021#issuecomment-1594976635) in the public domain and waive my rights regarding it; feel free to include it to whatever repository is suitable for you.

anntzer, 2025-08-01:
I don't think I have that tiff file anymore (IIRC I just took a single-frame MDA to generate this test nd file and immediately discarded the tiff after verifying the issue).