ome/bioformats #4022: Support for JPEG XL in a future release

MustafaArikan, 2023-06-16:
Are there plans for supporting the new JPEG standard (JPEG XL)?

It has a lot of features and could soon replace JPEG.

There is a reference implementation: https://github.com/libjxl/libjxl
Link: https://cloudinary.com/blog/the-case-for-jpeg-xl

dgault, 2023-06-19:
Hi @MustafaArikan, we currently have no immediate plans for adding support but it may be something we look at in the future. Are you actively using JPEG XL at the moment or are you looking to use it in the future?

MustafaArikan, 2023-06-24:
Hi @dgault, I am actually using it already. The quality is great. And the compression ratio is very high.
Would be great if there was support in Fiji through bioformats.