ome/bioformats #4024: BDV: Z downsampling not currently supported in viewers

dgault, 2023-06-22:
This issue was raised on forum thread https://forum.image.sc/t/omero-cli-import-of-bigdataviewer-file-throws-errors/82478/10
A sample file was provided by the user which reproduces the issue: https://zenodo.org/record/8060528

When opened with Bio-Formats directly using command line tools, the file opens without exceptions and there are 60 series (with the noflat option it is 15 series, but all open without exception), each can be opened and displayed without issue.

When imported into OMERO however the images do not appear to render in the web client, as reported by the user.  In the logs the below exceptions can be seen:

```
2023-06-22 10:37:01,876 INFO  [                 org.perf4j.TimingLogger] (.Server-10) start[1687430221876] time[0] tag[omero.call.exception]
2023-06-22 10:37:01,877 WARN  [        ome.services.util.ServiceHandler] (.Server-10) IllegalArgumentException thrown.

java.lang.IllegalArgumentException: Invalid Z index: 2/1
	at loci.formats.FormatTools.getIndex(FormatTools.java:473)
	at loci.formats.FormatTools.getIndex(FormatTools.java:408)
	at loci.formats.ChannelSeparator.getIndex(ChannelSeparator.java:264)
	at loci.formats.ReaderWrapper.getIndex(ReaderWrapper.java:473)
	at ome.io.bioformats.BfPixelsWrapper.getTile(BfPixelsWrapper.java:350)
	at ome.io.bioformats.BfPixelBuffer.getTile(BfPixelBuffer.java:499)
	at omeis.providers.re.data.PlaneFactory.createPlane(PlaneFactory.java:193)
	at omeis.providers.re.GreyScaleStrategy.renderAsPackedInt(GreyScaleStrategy.java:153)
	at omeis.providers.re.Renderer.renderAsPackedInt(Renderer.java:558)
	at ome.services.RenderingBean.renderAsPackedInt(RenderingBean.java:512)
	at ome.services.RenderingBean.renderCompressed(RenderingBean.java:542)
	at jdk.internal.reflect.GeneratedMethodAccessor2183.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
```

```
2023-06-22 10:37:00,596 INFO  [                 org.perf4j.TimingLogger] (.Server-14) start[1687430220511] time[84] tag[omero.call.exception]
2023-06-22 10:37:00,596 WARN  [        ome.services.util.ServiceHandler] (.Server-14) Unknown exception thrown.

java.lang.RuntimeException: ome.io.nio.DimensionsOutOfBoundsException: Z '2' greater than sizeZ '1'.
	at omeis.providers.re.data.PlaneFactory.createPlane(PlaneFactory.java:225)
	at omeis.providers.re.GreyScaleStrategy.renderAsPackedInt(GreyScaleStrategy.java:153)
	at omeis.providers.re.Renderer.renderAsPackedInt(Renderer.java:558)
	at ome.services.ThumbnailBean.createScaledImage(ThumbnailBean.java:648)
	at ome.services.ThumbnailBean._createThumbnail(ThumbnailBean.java:845)
	at ome.services.ThumbnailBean.retrieveThumbnail(ThumbnailBean.java:1189)
	at ome.services.ThumbnailBean.retrieveThumbnailAndUpdateMetadata(ThumbnailBean.java:1140)
	at ome.services.ThumbnailBean.getThumbnailByLongestSide(ThumbnailBean.java:1320)
	at jdk.internal.reflect.GeneratedMethodAccessor2199.invoke(Unknown Source)
```

I believe this is due to the image resolutions also being downsampled in Z. For each series the highest resolution has 5 Z levels, the next resolution has 2 Z levels and the 2 lowest resolutions have a single plane. Currently viewers such as the OMERO web viewer do not support Z downsampling and leads to issues as described here. 

It may be possible for the BDV reader to work around these issues by returning duplicate planes for the missing Z at lower resolutions.

NicoKiaru, 2023-06-27:
If you go for this, would it be doable to add (a method?) that returns the number of original Z planes (and their indices ?) for the viewers that can deal with Z subsampling ?

dgault, 2023-06-28:
Hi @NicoKiaru, we had some further discussion on this at our formats meeting this week and the consensus was that the readers are currently behaving as they should do. This particular issue was due to a limitation in OMERO and that is really were the long term solution should be. As such we ruled out implementing any work around or hack for the reader. I will close this issue as it is not  something we are going to change at a Bio-Formats level.

closed 2023-06-28 by dgault