ome/bioformats #4035: LIF reader Illegal argument exception

jburel, 2023-06-29:
```
stacktrace=java.lang.IllegalArgumentException
	at java.nio.Buffer.position(Buffer.java:244)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:650)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:494)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:482)
	at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:670)
```
Error using bf 6.3.1
https://www.openmicroscopy.org/qa2/qa/feedback/33438/

Close if no longer relevant

melissalinkert, 2023-06-29:
I can't log in to QA - was there more to the stack trace? In particular, we'd need to know the line number in the reader that this `skipBytes` call is coming from. Also don't see an uploaded file for 33438, so even with more context it may be difficult to say for sure if this was fixed.

jburel, 2023-06-29:
```
stacktrace=java.lang.IllegalArgumentException
	at java.nio.Buffer.position(Buffer.java:244)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:650)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:494)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:482)
	at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:670)
	at loci.formats.in.LIFReader.openBytes(LIFReader.java:352)
	at loci.formats.ImageReader.openBytes(ImageReader.java:461)
	at loci.formats.ChannelFiller.openBytes(ChannelFiller.java:156)
	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:229)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.MinMaxCalculator.openBytes(MinMaxCalculator.java:269)
	at ome.services.blitz.repo.ManagedImportRequestI.parseDataByPlane(ManagedImportRequestI.java:872)
	at 
```

dominikl, 2024-06-06:
Another similar one [QA 41692](https://www.openmicroscopy.org/qa2/qa/feedback/41692/).

melissalinkert, 2024-06-07:
@dominikl : I can't log in to QA, it looks like as with the previous QA report there is no file. Which version was QA 41692 reported against?

melissalinkert, 2025-05-22:
Closing, as there were no files associated with any of these reports. If we get another report with a file, feel free to re-open. Only other thought is that this looks similar to #3879, so may be a duplicate of that (but impossible to know for sure without more information).

closed 2025-05-22 by melissalinkert