ome/bioformats #4036: AFI / SVS: getMagnification results in NullPointerException

dgault, 2023-06-29:
This issue was raised on forum thread https://forum.image.sc/t/error-uploading-afi-image/81073

The original thread was due to failures with an afi file-set. However the exception appears to be in the underlying SVSReader and can be reproduced using Bio-Formats 6.13.0. The exception doesn't occur if you just read the SVS file directly, only as part of the AFI.

It looks as though the SVSReader initializes magnification to null as a Double, when the AFIReader calls getMagnification (returns a primitive double) and no magnification is set, then an NPE is thrown.

```
Exception in thread "main" java.lang.NullPointerException
	at loci.formats.in.SVSReader.getMagnification(SVSReader.java:635)
	at loci.formats.in.AFIReader.initFile(AFIReader.java:334)
	at loci.formats.FormatReader.setId(FormatReader.java:1466)
```

melissalinkert, 2023-06-29:
Is the problematic dataset available anywhere? I didn't see anything relevant under `inbox/`.

dgault, 2023-07-03:
Should now be available on scratch/inbox/imagesc-81073

PR #4081 (2023-10-13): SVS: fix a variety of small issues
c574360 fixes #3986. Most data is unaffected, but `CMU-1.svs` now has different behavior. Compare `showinf -noflat -omexml -series 1` and `showinf -noflat -omexml series 2` with and without this PR. Without this PR, a label image is shown the one named `macro image` and a macro image is shown for the one named `label image`. With this PR, the names and images should match. I think this behavior change warrants a minor release, but happy to hear other opinions.

1b6da10 trivially fixes #4036.

593c819 fixes #3743 by setting the X and Y position on the full-res [... cut]
commit 1b6da10d (2023-10-13): SVS: handle missing magnification

Fixes #4036
closed 2023-10-13 by dgault