ome/bioformats #4046: Build is broken when OME servers are down

aubreyrjones, 2023-07-12:
It appears building the bioformats library depends extensively on the server `https://artifacts.openmicroscopy.org`. This server appears highly unstable, and is currently down.

It would be better if builds did not depend on precompiled artifacts from a privately-managed server. Is it possible to use github or another publicly hosted artifact repository?

aubreyrjones, 2023-07-12:
The server connection may actually be some kind of networking issue. The servers seem up for colleagues in some locations.

melissalinkert, 2023-07-12:
@aubreyrjones, could you please let us know the build command you're using and the complete error message that you see? https://artifacts.openmicroscopy.org is working as we expect, and we have not seen recent build failures on our side that would suggest a problem with this server (see e.g. builds in https://github.com/ome/bioformats/actions).

aubreyrjones, 2023-07-13:
There is something very strange going on in fact, because I cannot even access the `openmicroscopy.org` website. I've contacted my ISP and they say there is no routing problem, and I've replaced my router with a different one to eliminate my own routing problems. All hosts on my home LAN act this way, while colleagues on other networks have no problems.

I'm running `mvn install`.

I get the following, ending in `connection refused`:

```
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for Bio-Formats projects 7.0.0-SNAPSHOT:
[INFO] 
[INFO] Bio-Formats projects ............................... SUCCESS [  0.486 s]
[INFO] libjpeg-turbo Java bindings ........................ SUCCESS [  0.320 s]
[INFO] Bio-Formats API .................................... SUCCESS [  1.039 s]
[INFO] BSD Bio-Formats readers and writers ................ FAILURE [01:01 min]
[INFO] Bio-Formats library ................................ SKIPPED
[INFO] Bio-Formats Plugins for ImageJ ..................... SKIPPED
[INFO] Bio-Formats command line tools ..................... SKIPPED
[INFO] bioformats_package bundle .......................... SKIPPED
[INFO] Bio-Formats testing framework ...................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:03 min
[INFO] Finished at: 2023-07-13T14:28:05-05:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project formats-bsd: Could not resolve dependencies for project ome:formats-bsd:jar:7.0.0-SNAPSHOT: Failed to collect dependencies at cisd:jhdf5:jar:19.04.0: Failed to read artifact descriptor for cisd:jhdf5:jar:19.04.0: Could not transfer artifact cisd:jhdf5:pom:19.04.0 from/to ome (https://artifacts.openmicroscopy.org/artifactory/maven/): Transfer failed for https://artifacts.openmicroscopy.org/artifactory/maven/cisd/jhdf5/19.04.0/jhdf5-19.04.0.pom: Connect to artifacts.openmicroscopy.org:443 [artifacts.openmicroscopy.org/134.36.65.239] failed: Connection refused (Connection refused) -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :formats-bsd
```

melissalinkert, 2025-03-17:
Closing as this appears to have been an isolated problem, but please feel free to re-open if it is still an issue. If it is still not possible to build, I expect we would need to know the source IP address in order to investigate any further.