ome/bioformats #4065: CellSens IndexOutOfBoundsException

jburel, 2023-08-04:
```
java.lang.IndexOutOfBoundsException: Index: 4, Size: 4
	at java.util.ArrayList.rangeCheck(ArrayList.java:659)
	at java.util.ArrayList.get(ArrayList.java:435)
	at loci.formats.in.CellSensReader.parseETSFile(CellSensReader.java:1210)
	at loci.formats.in.CellSensReader.initFile(CellSensReader.java:742)
	at loci.formats.FormatReader.setId(FormatReader.java:1443)
	at loci.formats.ImageReader.setId(ImageReader.java:849)
	at ome.formats.importer.OMEROWrapper$4.setId(OMEROWrapper.java:167)
```
Tested using omero.insight 5.8.1  during workshop in Tuscon

jburel, 2023-08-04:
I do have the files cc @dgault @melissalinkert

melissalinkert, 2023-08-04:
That looks a lot like #3944, which was fixed in 6.12.0. The stack trace suggests that an earlier version of Bio-Formats is being used, since the line numbers don't match up for 6.12.0:

https://github.com/ome/bioformats/blob/v6.12.0/components/formats-gpl/src/loci/formats/in/CellSensReader.java#L1210

That's a little odd though since OMERO.insight 5.8.1 has the Bio-Formats 6.12.0 jars.

jburel, 2023-08-04:
I can confirm that  I can open the images using the latest version of BioFormats