ome/bioformats #4066: Slidebook: ArrayIndexOutOfBoundsException

dgault, 2023-08-04:
Issue was reported on QA-33754, import failure during an OMERO workshop.

`Server name: workshop.openmicroscopy.org, version: 5.6.6`

The failure looks to be due to no pixelOffsets being found resulting in much of the metadata not being parsed. The stack trace is as below:
```
java.lang.ArrayIndexOutOfBoundsException: 0
	at loci.formats.in.SlidebookReader.initFile(SlidebookReader.java:634)
	at loci.formats.FormatReader.setId(FormatReader.java:1443)
	at loci.formats.ImageReader.setId(ImageReader.java:849)
	at ome.formats.importer.OMEROWrapper$4.setId(OMEROWrapper.java:167)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelFiller.setId(ChannelFiller.java:234)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:293)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.Memoizer.setId(Memoizer.java:662)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
```

jburel, 2023-08-04:
I will speak to the person today so e can get the file

stephenshank, 2023-12-01:
I am also experiencing this issue on 7.0.1, trying to interface with the Java library using the documentation.

dgault, 2023-12-01:
@stephenshank, do you have a sample file that we could use to reproduce the exception? If you need a suitable upload location then we recommend using https://zenodo.org/

stephenshank, 2023-12-04:
I have shared data via email. For what it's worth, the standard output reports finding over two hundred pixel offsets, but the script still fails at the same exact location. The bug is also rather pervasive, occurring in all six datasets that I possess.

Thank you so much for looking into this! I will try to convince my collaborator to share some data publicly on Zenodo, to keep with the spirit of openness of the project.

leonardblaschek, 2024-11-27:
Any update on this? I'm running into similar errors on fedora linux. 

bfconvert (v. 8.0.1) error:

```
Finding offsets to pixel data
Determining dimensions
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0
        at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:100)
        at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:106)
        at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:302)
        at java.base/java.util.Objects.checkIndex(Objects.java:385)
        at java.base/java.util.ArrayList.get(ArrayList.java:427)
        at loci.formats.in.SlidebookReader.initFile(SlidebookReader.java:654)
        at loci.formats.FormatReader.setId(FormatReader.java:1480)
        at loci.formats.ImageReader.setId(ImageReader.java:864)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:562)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1356)
```

Fiji Bio-Formats (v. 8.0.0-20241020.011418-33 or 8.0.1) error:

```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
  at java.util.ArrayList.rangeCheck(ArrayList.java:659)
  at java.util.ArrayList.get(ArrayList.java:435)
  at loci.formats.in.SlidebookReader.initFile(SlidebookReader.java:654)
  at loci.formats.FormatReader.setId(FormatReader.java:1480)
  at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:498)
  at loci.plugins.in.ImportProcess.execute(ImportProcess.java:141)
  at loci.plugins.in.Importer.showDialogs(Importer.java:156)
  at loci.plugins.in.Importer.run(Importer.java:77)
  at loci.plugins.LociImporter.run(LociImporter.java:78)
  at ij.IJ.runUserPlugIn(IJ.java:244)
  at ij.IJ.runPlugIn(IJ.java:210)
  at ij.Executer.runCommand(Executer.java:152)
  at ij.Executer.run(Executer.java:70)
  at java.lang.Thread.run(Thread.java:750)
```

The same files can be opened on a colleagues Windows machine running Bio-Formats v 8.0.1.