ome/bioformats #4087: ZeissCZIReader get lut error

SinnoSong, 2023-08-24:
I have a `czi` file , read color of one channel , the color is `#FF0000FF`, in the source code use int to parse this number.

joshmoore, 2023-08-24:
Hi @SinnoSong. Sorry, I'm not understanding your question. Can you share more about what you're trying to do? What tool are you using? Or are you writing your own code? If so, can you share it? What is the entire error that you are getting?

SinnoSong, 2023-08-24:
@joshmoore Sorry, I use `formats-gpl` to read a `czi` file. The file has 2 channels, the color of a channel is `#FF0000FF`. When I use `get16BitLookupTable` method to get Lut, the method throws `NumberFormatException`. I see the `get16BitLookupTable` method use `int colorValue = Integer.parseInt(color, 16);` to convert `FF0000FF` to a integer number, `FF0000FF` is greater than Integer max value.

PR #4088 (2023-10-03): CZI: unify channel color parsing
Fixes #4087.

`get*LookupTable` now parses colors using the same logic that is used when setting `Channel.Color`. This allows for both RGB and ARGB colors; if specified, the alpha is ignored. If anything other than '#' is removed from the original color string, the change will be logged at `DEBUG`.
commit 7b51510e (2023-10-03): CZI: unify channel color parsing

Fixes #4087.

get*LookupTable now parses colors using the same logic that is used
when setting Channel.Color. This allows for both RGB and ARGB colors;
if specified, the alpha is ignored. If anything other than '#' is removed
from the original color string, the change will be logged at DEBUG.