ome/bioformats #4089: Loading 64-bit int TIFFs

edwardando, 2023-08-28:
Dear all,
As a Fiji user I'm a long time user and appreciator of BioFormats.
However... there appears to be quite dangerous behaviour when loading 64-bit int or uint tiffs, as I posted here:

https://forum.image.sc/t/saving-and-loading-64-bit-int-tiffs/85523

@cgohlke proposes a very reasonable solution, and I'd be interested in your opinion

dgault, 2023-08-30:
Thanks @edwardando for opening the issue, hopefully this will be something we can improve in a future release, the suggested fix (copied below) seems like a sensible option in this case. 

```
The fix will probably be to throw an UnknownFormatException if BitsPerSample==64 and SampleFormat!=3.
```

edwardando, 2023-08-31:
Fully agreed, thanks!

PR #4098 (2023-09-25): TIFF: throw exception on 64-bit int data
Fixes #4089. Test file in `inbox/gh-4089`, generated as described in https://forum.image.sc/t/saving-and-loading-64-bit-int-tiffs/85523/2.

Without this PR, `showinf im-i8.tif` should display an image with `double` type. `tiffinfo` or `tiffdump` should confirm that there 64 bits per sample, but with a `SampleFormat` of 2 indicating signed int data.

With this PR, `showinf -debug im-i8.tif` should throw `FormatException` indicating that the pixel type is unsupported.

We have no real way of supporting this data in Bio-Formats right now, since the OME mod [... cut]
closed 2023-09-25 by dgault