ome/bioformats #4102: CZI: Autostich fails

dgault, 2023-10-10:
Issue was raised on forum thread https://forum.image.sc/t/zoom-from-overview-to-detailed-scan-for-imported-czi-files/85002/6 

A sample was provided that reproduces the issue with Bio-Formats 7.0.0 - https://zenodo.org/record/8423633

When opening the file, even with the autostitch option selected, the file opens as 36 individual series. This appears to be 5 tiles, most of which have 7 resolutions, but 1 has 6 resolutions, the sizes of these 6 resolutions do not match the other 4. There is also an additional label and macro attachment image.

When using the noflat option it is returning 7 series as below:
```
Series count = 7
Series #0 :
	Resolutions = 7
		sizeX[0] = 33309
		sizeX[1] = 16654
		sizeX[2] = 8327
		sizeX[3] = 4163
		sizeX[4] = 2081
		sizeX[5] = 1040
		sizeX[6] = 520
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 33309
	Height = 29285
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 128 x 112
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #1 :
	Resolutions = 7
		sizeX[0] = 33357
		sizeX[1] = 16678
		sizeX[2] = 8339
		sizeX[3] = 4169
		sizeX[4] = 2084
		sizeX[5] = 1042
		sizeX[6] = 521
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 33357
	Height = 29295
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 128 x 112
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #2 :
	Resolutions = 7
		sizeX[0] = 33331
		sizeX[1] = 16665
		sizeX[2] = 8332
		sizeX[3] = 4166
		sizeX[4] = 2083
		sizeX[5] = 1041
		sizeX[6] = 520
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 33331
	Height = 20698
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 128 x 79
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #3 :
	Resolutions = 6
		sizeX[0] = 26093
		sizeX[1] = 13046
		sizeX[2] = 6523
		sizeX[3] = 3261
		sizeX[4] = 1630
		sizeX[5] = 815
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 26093
	Height = 5576
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 128 x 27
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #4 :
	Resolutions = 7
		sizeX[0] = 34784
		sizeX[1] = 17392
		sizeX[2] = 8696
		sizeX[3] = 4348
		sizeX[4] = 2174
		sizeX[5] = 1087
		sizeX[6] = 543
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 34784
	Height = 24965
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 128 x 91
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #5 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 698
	Height = 649
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 12 x 9
	Thumbnail size = 128 x 119
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 12
	Metadata complete = false
	Thumbnail series = true
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #6 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 1458
	Height = 777
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1458 x 119
	Thumbnail size = 128 x 68
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 12
	Metadata complete = false
	Thumbnail series = true
	-----
	Plane #0 <=> Z 0, C 0, T 0
```

dgault, 2023-10-20:
Another auto-stitching issue has been reported on forum thread https://forum.image.sc/t/pb-when-i-open-czi-stitching-image/87585/2

A sample image was provided and is available at `inbox/imagesc-87585` and the issue can be reproduced using Bio-Formats 7.0.1 with the CZI auto-stitching option enabled.

In this particular case the user was able to save a stitched image with Zen 3.4 and that opens as expected, but when saved with Zen 3.8 the image appears "distorted" with the stitching option enabled. In QuPath zooming in to the highest resolution level produced a normal looking image, similarly cropping regions of the highest resolution also seemed to return correct looking pixel data. When the stitching option is disabled the individual tiles appear to be correct and display as expected.

dgault, 2024-02-07:
The issue reported in https://github.com/ome/bioformats/issues/4102#issuecomment-1772752535 appears to be different to that in the original Issue description. I am splitting this report into a new seperate Issue https://github.com/ome/bioformats/issues/4150 which is addressed by https://github.com/ome/bioformats/pull/4138

MichaelDausmann, 2024-02-07:
I have done additional research on this issue.  

TLDR, I don't think the AutoStitch option is stitching scenes together.

a ‘Scene’ in CZI is represented by the ‘S’ Dimension. according to the [documentation](https://zeiss.github.io/libczi/imagedocumentconcept.html#autotoc_md6),

> And we have the letter ‘S’ in use, and it is used in the following way: sub-blocks with the same S-index form a set called “scene”. A scene is a rectangular (and axis aligned) region, and much like the bounding box, it is determined by taking all planes into consideration.

In the ZeissCZIReader, as best as i can tell, the ‘S’ dimension is being interpreted as a ‘Series’ dimension.

For context, I have attached the output of the CZIcmd utility for my test file...
[stitching_testfile_czicmdoutput.txt](https://github.com/ome/bioformats/files/14202671/stitching_testfile_czicmdoutput.txt)
Note that the file is arranged into 5 Scenes, each with their own pyramids.

I have also attached a detailed dump of all the subblocks in my test file...
[stitching_subblocks_detail.txt](https://github.com/ome/bioformats/files/14202646/stitching_subblocks_detail.txt)

I also have been able to debug through ZeissCZIReader.java as it reads my file
[image]

I noticed that despite autostitching being on by default, the sub-blocks in Scenes > 0 are being placed into seperate series e.g. block at i=489 ends up in Series 28.

Whether Autostitch ignoring scenes or the interpretation of 'S' as a 'Series' rather than a 'Scene' was intentional, I can't tell.  

I guess what I am asking is either through an additional option or by default, the 'S' Dimension should be ignored from a Series perspective so that all sub-blocks in my file with the same scale (zoom, z) end up in the same series.  that would mean for my test file I would end up with 9 series rather than 36 i.e. 7 series with all of the tiles from the pyramids, and one series each for the slide header and the macro view. 

I think as a 'feature', this would mean that a multi-scene .czi file would be imported e.g into Omero as a 'whole slide' image rather than an image per scene.

Hope this makes sense

MichaelDausmann, 2024-06-07:
@dgault  recommend closure of this ticket in favour of https://github.com/ome/bioformats/issues/4194